SpinalHDL/VexRiscv #238: single step fails during on chip debug when prediction configured to DYNAMIC or DYNAMIC_TARGET

syibin, 2022-02-22:
single step fails during on chip debug when prediction configured to DYNAMIC or DYNAMIC_TARGET

I met an issue related to prediction. I setup the gdb + openocd + Briey SOC co-simulation as guided by Vex readme. With default Briey which use STATIC prediction, the hello world project works fine - can print out "hello" string. However, if I switch the prediction to DYNAMIC, and rerun. no string will be printed.
Meanwhile, I also had a try on board. with prediction to STATIC, everything works fine with sdk. However, if prediction switch to DYNAMIC or DYNAMIC TARGET, soft/harden breakpoint still works well. While single step will have issue - when i put si in gdb, sometimes, next 1 instruction will be run, sometimes next 1 instruction will be skipped unexpectedly and the next next one will be run instead.

Steps:
0. I download the latest VexRiscv and openocd_riscv about 1 month ago.
1. Generate Briey - sbt "runMain vexriscv.demo.Briey".
For the working SOC, I just use the default option; for the failing SOC, I made following changes:
   a) prediction set to DYNAMIC or DYNAMIC_TARGET for IBusCachedPlugin
   b) wayCount set to 2, twoCycleRam set to false for InstrunctionCacheConfig (these change won't cause issue, just to clarify my changes)
   c) regFileReadyKind set to ASYNC for RegFilePlugin (to compatible with dynamic prediction)
2. cd src/test/cpp/briey, and run "make clean run"
3. launch openocd: 
src/openocd -f tcl/interface/jtag_tcp.cfg -c "set BRIEY_CPU0_YAML /home/syibin/sda3/VexRiscv-master/cpu0.yaml" -f tcl/target/briey.cfg
4. compile uart project by running makefile in VexRiscvSocSoftware-master/projects/briey/uart
5. launch gdb with compiled uart.elf:
~/sda3/riscv32-unknown-elf.gcc-10.2.0.rv32i.ilp32.newlib/bin/riscv32-unknown-elf-gdb uart.elf
6. run following commands in gdb:
target remote localhost:3333
set remotetimeout 60
set arch riscv:rv32
monitor reset halt
load
b main
continue
si
si
c

SW version:
I think the gcc/openocd/gdb's version won't matter since I tried two different versions of openocd and gdb, the result keep unchanged.
For Verilator: Verilator 4.040 2020-08-15 rev v4.040

Dolu1990, 2022-02-22:
So part of the issue was the simulation testbench which had a floating uart rxd. It is now fixed.

Dolu1990, 2022-02-22:
About 

> While single step will have issue - when i put si in gdb, sometimes, next 1 instruction will be run, sometimes next 1 instruction will be skipped unexpectedly and the next next one will be run instead.

So, you are using "si" which is single instruction. (not single step).

Using "si" i observe : 

```
(gdb) si
36		reg->CLOCK_DIVIDER = config->clockDivider;
(gdb) si
36		reg->CLOCK_DIVIDER = config->clockDivider;
(gdb) si
0x40000144	36		reg->CLOCK_DIVIDER = config->clockDivider;
(gdb) si
0x40000144	36		reg->CLOCK_DIVIDER = config->clockDivider;
(gdb) si
36		reg->CLOCK_DIVIDER = config->clockDivider;
(gdb) si
0x4000014c	36		reg->CLOCK_DIVIDER = config->clockDivider;
```

which is a example of the symptomes you have seen right ?

syibin, 2022-02-23:
Hi, Dolu

yes, the simulation difference between STATIC and DYNAMIC is caused by the floating uart rx. After the fix, the DYNAMIC mode can also see the "Hello" string. Thanks for the fix - otherwise, I may go to the wrong direction. :)

Regarding board level difference, I'm still working on to figure out why, will keep you updated once I find something.

Thanks
Yibin

Dolu1990, 2022-02-24:
One possibility is that in the sim the dynamic predictors are initialized with some stuff, while in sim it isn't ?

syibin, 2022-05-31:
Hi, Dolu

The issue is fixed by the your recent fix for https://github.com/SpinalHDL/VexRiscv/issues/254. 

Thanks again for the support. :)

Thanks
Yibin

closed 2022-05-31 by Dolu1990