SpinalHDL/VexRiscv #241: murax code simulation

m-pp, 2022-04-01:
Hi All,

I am trying to simulate a c code on murax using verilator. I want to get the cycle count after the code run is complete. I use this: sbt "test:runMain vexriscv.MuraxSim"

Currently, the hello world code runs through and prints the custom string. But, waits for an interrupt and never prints the following statement in VexRiscv-Root/src/test/cpp/murax

"Had simulate " << workspaceCycles << " clock cycles in " << duration*1e-9 << " s (" << workspaceCycles / (duration*1e-9) << " Khz)" << endl

Is that a bug? How can I get the run-time cycle count for hello_world.c?

Dolu1990, 2022-04-04:
Hi ^^

So,  'sbt "test:runMain vexriscv.MuraxSim"'  isn't using src/test/cpp/murax. There is curently to different test environnement : 
- raw C++/Verilator via src/test/cpp/murax
- Scala / SpinalSim / Verilator which is 'sbt "test:runMain vexriscv.MuraxSim"' 

> Is that a bug? How can I get the run-time cycle count for hello_world.c?

So either you can use the cpp sim : 
Then go in src/test/cpp/murax and run the simulation with:
make clean run

Either the way you were actualy doing, but you will need to extends the src/test/scala/vexriscv/MuraxSim.scala testbench to print it.

m-pp, 2022-04-04:
OK, thanks.

When I use the cpp sim with make clean run, with gdb hello_world.elf,  the simulation shows the following:

BOOT
CONNECTED
hello world

And stays there forever. Is it waiting for an interrupt? 

It looks like the "MuraxWorkspace().run(100e6); in src/test/cpp/murax main.cpp doesnt quit?

Dolu1990, 2022-04-07:
> And stays there forever. Is it waiting for an interrupt?

I think that's normal, it is just done after it, no interrupt.

> run(100e6)

That is a very long time ^^
try reduce it to 100e3

m-pp, 2022-04-07:
No, that doesnt help. The crt.S has an infinite loop.

Dolu1990, 2022-04-08:
Yes, it is expected that the CPU endup in the infinit loop of crt.S after the main is executed.

In other words, it isn't expected that the sim stop by itself after the execution of the software, it wasn't implemented for that.

Or the CPU do not reach it ?

m-pp, 2022-04-08:
Agree, but MuraxWorkspace().run() does not exit.

Dolu1990, 2022-04-08:
Ahhhh i can see now, the timeout parameter is never used in https://github.com/SpinalHDL/VexRiscv/blob/master/src/test/cpp/common/framework.h#L183

That's why. So fixing that, and MuraxWorkspace().run(100e3); should make things behave right.

m-pp, 2022-04-11:
No, that doesnt help. MuraxWorkspace().run(100e3) also does not exit. 

Commenting out MuraxWorkspace().run(100e3)  prints the timers and exits main in src/test/cpp/murax/main.cpp

Dolu1990, 2022-04-11:
I pused a fix in dev, where the timeout is not specifyed in simulated seconds. The sim end when it is reached.

m-pp, 2022-04-12:
Thanks. This works fine for the scala simulation with sbt "test:runMain vexriscv.MuraxSim"

But, with the cpp make clean run in src/test/cpp/murax/, the code does not exit, still.

m-pp, 2022-04-12:
Maybe I am doing something wrong. 

This is what I want to do: measure cycle count for a custom code on different vexriscv configurations (pipeline stages, FPU, custom instructions etc). What is the best way to do this?

Dolu1990, 2022-04-13:
Ahhh
So, the best way to do it is in software, else you will mesure all sort of other things at once.

You can use the following for printing stuff : 
https://github.com/SpinalHDL/VexRiscvSocSoftware/blob/master/projects/murax/dhrystone/src/stdlib.c#L62

The following code to init the timerA counting up :
https://github.com/SpinalHDL/VexRiscvSocSoftware/blob/master/projects/murax/dhrystone/src/main.c#L19

And this to read the timer : 
https://github.com/SpinalHDL/VexRiscvSocSoftware/blob/master/projects/murax/dhrystone/src/main.c#L38

Dolu1990, 2022-04-13:
Else what you can do, is extends the testbench with white box probing, to check which PC is executed, and start counting things, that what i have done for NaxRiscv, but that can be more tricky.

m-pp, 2022-04-25:
Thanks. 

> Ahhh So, the best way to do it is in software, else you will mesure all sort of other things at once.
> 
Thanks. Did you mean using simulation (C++ or sbt) or with the board?

Dolu1990, 2022-04-25:
both ^^
As long as the test isn't too long in simulation, i would say, go for the simulation.

m-pp, 2022-04-26:
> I pused a fix in dev, where the timeout is not specifyed in simulated seconds. The sim end when it is reached.

I dont think this was pushed to the master branch. @Dolu1990

Dolu1990, 2022-04-26:
Merged in master

closed 2022-04-26 by Dolu1990