SpinalHDL/VexRiscv #244: Adding FPU

mattvlsi, 2022-04-05:
Hi,

I want to generate VexRiscv and Briey with Double FPU plugin. I'm a newby in SpinalHDL system. How can I do this? Thanks.

Dolu1990, 2022-04-07:
Hi,

So, as a first step i would say the best is with the Float FPU, (because the double FPU require 64 bits bus, which would require a few more modification in Briey).

So for Float FPU, you need to add the FpuPlugin to the VexRiscv config, and that's it.

So, for instance after https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/Briey.scala#L133

```scala
new FpuPlugin(
  externalFpu = false,
  p = FpuParameter(
    withDouble = false
  )
)
```

ashuthosh-mr, 2024-03-01:
Also, on the hardware side of the story,

```new FpuPlugin(
  externalFpu = false,
  p = FpuParameter(
    withDouble = true
  )
)
```

I made withDouble = true. Unable to generate the respective GenFull, what other changes might be needed in the vexriscv scala to generate the GenFull with rv32imafdc?

Dolu1990, 2024-03-04:
I would say, start first with 32 bits floating point only, to not add the additional 64 bits memdatawidth bus complication.

> However, when I use -mabi=ilp32d -march=rv32imafdc, it compiles but do not simulate on a GenFull vexriscv with FPU as it gets stucks in the simulation.

When you run the simulation with 64 bits mem data width, you also need to specify a few define to the make command, have you done so ? else The simulation will assume 32 bits data bus