SpinalHDL/VexRiscv #251: externalResetVector infers latches

ejeffrey, 2022-05-10:
I'm working on an application where I would like to use a run-time programmable reset vector.  I modified GenFulNoMmuNoCache.scala to set the `IBusSimplePlug(resetVector = null, ...)` which seems to be intended to allow an external input for the reset vector.  However, the generated verilog unsurprisingly looks like this:

```
always @(posedge clk or posedge reset) begin
    if (reset) begin
        pcReg <= externalResetVector;
    end else begin
        pcReg <= next_pc;
    end
end
```

If `externalResetVector`is assigned to a non-constant value this represents an asynchronous load rather than reset and quartus gives a warning message about emulating this with latches.  

The impact to fmax is negligible and I'm pretty sure I don't care much about the particular opportunities for simulation / synthesis mismatch that could come from this, but I also have heard about where good intentions lead.  If I were writing this in native verilog I could convert this to an explicit latch or use a synchronous load with a possible performance cost. 

I guess my questions are:

1) Is this an anticipated use case, or is externalResetVector only intended to be used with synthesis time constant values?
2) Should I just live with this behavior or is there a better way to do this?
3) Is there anything that is probably going to come back to bite me if I run with this?

Dolu1990, 2022-05-10:
> Is this an anticipated use case, or is externalResetVector only intended to be used with synthesis time constant values?

It isn't very much a anticipaged use case. It could be easily fixed with an additional option. and yes it wasmostly intended as a constant value.
One easy workaround would be to generate the CPU with a syncronous reset. Would that be good for you ?

> Should I just live with this behavior or is there a better way to do this?

I would say let's fix it ^^

> Is there anything that is probably going to come back to bite me if I run with this?

I would not play with the devil :)

ejeffrey, 2022-05-10:
Yes, I think a synchronous reset would work for me.

Dolu1990, 2022-05-11:
Do you know how to setup that configuration

ejeffrey, 2022-05-12:
I was able to do this by modifying the generates verilog to verify that it works as expected.  The first time I tried it seemed to reduce Fmax by about 15% but that must have been an issue with my top level design and I was able to get it to work without penalty at around 190 MHz.  I would like to be able to make this change in the spinalhdl configuration but I don't see how to do that.

Dolu1990, 2022-05-12:
Hmmm 
replacing : 
SpinalVerilog(cpu())

By :
SpinalConfig(defaultConfigForClockDomains = ClockDomainConfig(resetKind = SYNC))(cpu())

Should do the trick

ejeffrey, 2022-05-12:
That did the trick, thanks!

closed 2022-05-13 by Dolu1990