SpinalHDL/VexRiscv #252: Strange uart print when enabling cache and compressed instructions.

hqd7777, 2022-05-11:
Hi, I find a very strange issue when I play with the Briey SOC with cache and C. The software project I use is the UART example in VexRiscvSocSoftware and I also compile it with compressed instructions enabled. When I debug it using riscvgdb and riscv openocd, I find printing error only if I set a breakpoint before the loop and then debug step by step. For example I set breakpoint at line 21
	print("Hello !\n");
It will print "Hello!" correctly and change the line. However, the characters after that will be messed up. Only when you use continue other than step can make the output correct after few outputs. 
[image]

By the way any breakpoints inside the loop nor will not cause that error.

Dolu1990, 2022-05-11:
Hooo that's a very interresting bug. It was on hardware or in simulation ?

Dolu1990, 2022-05-11:
Could you send me the exact omdification made to briey ? (especialy as you enabled RVC)

Dolu1990, 2022-05-11:
Got the issue reproduced in simulation <3
I'm on it.

hqd7777, 2022-05-11:
> On my side this fix the issue : [4fff62d](https://github.com/SpinalHDL/VexRiscv/commit/4fff62d3feeb05edcea46e66b25de0ecf3f274fa)
> 
> Let's me know if things aren't going well on your side ^^

Wow,I am going to check it on my simulation. Thank you very much!

hqd7777, 2022-05-11:
> On my side this fix the issue : [4fff62d](https://github.com/SpinalHDL/VexRiscv/commit/4fff62d3feeb05edcea46e66b25de0ecf3f274fa)
> 
> Let's me know if things aren't going well on your side ^^

It seems to work on my side too. I really appreciate for your help!

Dolu1990, 2022-05-12:
cool thanks ^^

closed 2022-05-12 by Dolu1990