SpinalHDL/VexRiscv #254: Similar debug issue with briey soc

hqd7777, 2022-05-17:
Hi, 
Similar to previous issue[https://github.com/SpinalHDL/VexRiscv/issues/252], I also find an issue in simulation when prediction is dynamic target. 
1. Openocd will crash using step by step in the while loop. I forget how I exactly make it, but it causes a core dumped using gdb step.
[image]
2. Similar to my previous ask, when I set a breakpoint at line 21 in main.c and then using step by step debug, it seems to be stuck at some point.
[image]

hqd7777, 2022-05-17:
If I use the stepi at the beginning, it seems that I will be trapped in the bss loop and never jump out

hqd7777, 2022-05-17:
For the first issue, it seems to jump to an incorrect address at line 32 of uart.h. I suppose that 0x3a is invalid for this one right?

Dolu1990, 2022-05-17:
Ahhhh damned
And it was with the exact same setup, (briey soc, only branch prediction changed + RVC)  ?

> 0x3a

Which 0x3a ?

hqd7777, 2022-05-17:
> Ahhhh damned And it was with the exact same setup, (briey soc, only branch prediction changed + RVC) ?
> 
> > 0x3a
> 
> Which 0x3a ?

Yes, I just changed the branch prediction. 
I mean when you use gdb stepi command at line 32, it seems to read instruction from address 0x0000003a.

hqd7777, 2022-05-17:
Actually the program is still running. But only the first letter will print and show on the screen. That might not be prediction error.

Dolu1990, 2022-05-17:
> step by step

So not instruction step by step, but C step by step right ?
Step in / step out / regular step ? can you describe exactly the scenario ?

Dolu1990, 2022-05-17:
Ahhh i reproduced something, i'm on it ^^

Dolu1990, 2022-05-17:
I just pushed another fix, i hope now all bugs are fixed ^^
Let's me know if you hit anything weird.

hqd7777, 2022-05-18:
> I just pushed another fix, i hope now all bugs are fixed ^^ Let's me know if you hit anything weird.

Thank you Charles. I am living in China and therefore not able to reply your message on time. Apologyze for unprecise descriptions that make you confused. I appreciate for your help to fix those issues.

Dolu1990, 2022-05-18:
>  I am living in China and therefore not able to reply your message promptly.  ApologyzeC for unprecise descriptions that make you confused.

No worries ^^

> It also fix the bugs on my side. I suppose you may close this issue.

Nice !
Thanks for the bug reporting :)

closed 2022-05-18 by Dolu1990