SpinalHDL/VexRiscv #255: [error] java.lang.ClassNotFoundException: vexriscv.demo.GenFull

mjchipguru, 2022-06-22:
When I ran this command, sbt "runMain vexriscv.demo.GenFull", the following error occurred. Please help. Thanks! 

mji@XPS-8930-5:~/VexRiscv/verilator$ sbt "runMain vexriscv.demo.GenFull"
[info] welcome to sbt 1.6.2 (Private Build Java 11.0.15)
[info] loading project definition from /home/mji/VexRiscv/verilator/project
[info] set current project to verilator (in build file:/home/mji/VexRiscv/verilator/)
[info] running vexriscv.demo.GenFull 
[error] java.lang.ClassNotFoundException: vexriscv.demo.GenFull
[error] 	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:476)
[error] stack trace is suppressed; run last Compile / runMain for the full output
[error] (Compile / runMain) java.lang.ClassNotFoundException: vexriscv.demo.GenFull
[error] Total time: 0 s, completed Jun 22, 2022, 2:08:52 PM

Dolu1990, 2022-06-23:
Hi,

You need to run the sbt command in the VexRiscv folder, not VexRiscv/verilator one ^^

mjchipguru, 2022-06-23:
I did try running this sbt at the parent directory and got the same error as shown below. After I did all the steps at the dependencies, it appears I could not find these files under src/ There is no main subdirectory under src/. 
src/main/scala/vexriscv/demo/GenFull.scala
src/main/scala/vexriscv/demo/GenSmallest.scala

mji@XPS-8930-5:~/VexRiscv$ sbt "runMain vexriscv.demo.GenFull"
[info] welcome to sbt 1.6.2 (Private Build Java 11.0.15)
[info] loading project definition from /home/mji/VexRiscv/project
[info] set current project to vexriscv (in build file:/home/mji/VexRiscv/)
[info] running vexriscv.demo.GenFull 
[error] java.lang.ClassNotFoundException: vexriscv.demo.GenFull
[error] 	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:476)
[error] stack trace is suppressed; run last Compile / runMain for the full output
[error] (Compile / runMain) java.lang.ClassNotFoundException: vexriscv.demo.GenFull
[error] Total time: 0 s, completed Jun 23, 2022, 8:47:48 AM
mji@XPS-8930-5:~/VexRiscv$

Dolu1990, 2022-06-27:
> I don't see the folder tree under VexRiscv/ .
> I followed all the steps under Dependencies:
> https://github.com/SpinalHDL/VexRiscv#dependencies
> Do I need to run any other extra step? Thanks.

Ahhh, git clone https://github.com/SpinalHDL/VexRiscv.git  && cd VexRiscv
:) ?