SpinalHDL/VexRiscv #257: Briey compilation error

everetto89, 2022-07-14:
Hi, 
I want to generate the standard Briey core with and without cached instruction and data buses. Later on I want to add the FPU plugin. 
At first when I try to generate the briey core without any modifications I get the following error: 

`[error] /home/user/Rackstation_Home/RISC-V/frameworks/Vexriscv/VexRiscv-Git/src/main/scala/vexriscv/TestsWorkspace.scala:36:17: not enough arguments for method apply: (gen: => T)spinal.core.SpinalReport[T] in class SpinalConfig.
[error] Unspecified value parameter gen.
[error]     SpinalConfig().generateVerilog {
[error]                 ^
[error] /home/user/Rackstation_Home/RISC-V/frameworks/Vexriscv/VexRiscv-Git/src/main/scala/vexriscv/demo/Briey.scala:399:18: reference to SpinalConfig is ambiguous;
[error] it is imported twice in the same scope by
[error] import spinal.core._
[error] and import vexriscv._
[error]     val config = SpinalConfig()
[error] `

Can anyone please help here? 

Further question: is it possible to integrate the FPU Plugin in a design with Non-Cached Buses? Thanks in advance!

Dolu1990, 2022-07-18:
Also, one question is, on which OS are you trying ?

everetto89, 2022-07-22:
Hi, 
after I recloned the repository the compilation of the standard configuration works fine without any issues. I'm using Ubuntu 20.04 LTS.
When i try to compile now with the DBusSimplePlugin instead of the DBusCachedPlugin everything works also fine.
But when I try to compile the Briey with the following IBusSimplePlugin

```
        new IBusSimplePlugin(
          resetVector = 0x80000000l,
          cmdForkOnSecondStage = true,
          cmdForkPersistence = false, //Required by the Xip controller
          prediction = NONE,
          catchAccessFault = false,
          compressedGen = false,
          bigEndian = false
        ),
```
I get the following error
```
[error] Exception in thread "main" java.lang.AssertionError: assertion failed
[error] 	at scala.Predef$.assert(Predef.scala:156)
[error] 	at spinal.core.package$.assert(core.scala:451)
[error] 	at vexriscv.plugin.IBusSimpleBus.toAxi4ReadOnly(IBusSimplePlugin.scala:110)
[error] 	at vexriscv.demo.Briey$$anon$3$$anon$4$$anonfun$7.apply(Briey.scala:313)
[error] 	at vexriscv.demo.Briey$$anon$3$$anon$4$$anonfun$7.apply(Briey.scala:312)
[error] 	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error] 	at vexriscv.demo.Briey$$anon$3$$anon$4.<init>(Briey.scala:312)
[error] 	at vexriscv.demo.Briey$$anon$3.<init>(Briey.scala:304)
[error] 	at vexriscv.demo.Briey.<init>(Briey.scala:254)
[error] 	at vexriscv.demo.Briey_UzL$$anonfun$main$1.apply(Briey.scala:509)
[error] 	at vexriscv.demo.Briey_UzL$$anonfun$main$1.apply(Briey.scala:508)
[error] 	at spinal.core.internals.PhaseCreateComponent$$anonfun$impl$62.apply(Phase.scala:2350)
[error] 	at spinal.core.fiber.Engine$$anonfun$create$1.apply$mcV$sp(AsyncCtrl.scala:147)
[error] 	at spinal.core.fiber.AsyncThread$$anonfun$1.apply$mcV$sp(AsyncThread.scala:59)
[error] 	at spinal.core.fiber.EngineContext$$anonfun$newJvmThread$1.apply$mcV$sp(AsyncCtrl.scala:39)
[error] 	at spinal.sim.JvmThread.run(SimManager.scala:51)
[error] Nonzero exit code returned from runner: 1
```
Can you please help here? 
Thanks in advance!

Dolu1990, 2022-07-22:
L110 =>     assert(plugin.cmdForkPersistence)

You need to enable the cmdForkPersistence option in the IBusSimplePlugin ^^

everetto89, 2022-07-22:
Thank you that was really helpful, now it works! 
One last issue: 
When i try to integrate the FPU Plugin with the following configuration
```
new FpuPlugin(
        externalFpu = false,
        p = new FpuParameter(
                withDouble = false
        )
),
```
with the DBusSimplePlugin of the following configuration

```
new DBusSimplePlugin(
        catchAddressMisaligned = true,
        catchAccessFault = true
),
```
I get the following error

```
[error] Exception in thread "main" java.lang.AssertionError: assertion failed: ??? vexriscv.plugin.DBusEncodingService
[error] 	at scala.Predef$.assert(Predef.scala:170)
[error] 	at spinal.core.package$.assert(core.scala:454)
[error] 	at vexriscv.Pipeline$class.service(Pipeline.scala:26)
[error] 	at vexriscv.VexRiscv.service(VexRiscv.scala:123)
[error] 	at vexriscv.plugin.FpuPlugin.setup(FpuPlugin.scala:157)
[error] 	at vexriscv.plugin.FpuPlugin.setup(FpuPlugin.scala:12)
[error] 	at vexriscv.Pipeline$$anonfun$build$2.apply(Pipeline.scala:47)
[error] 	at vexriscv.Pipeline$$anonfun$build$2.apply(Pipeline.scala:47)
[error] 	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error] 	at vexriscv.Pipeline$class.build(Pipeline.scala:47)
[error] 	at vexriscv.VexRiscv.build(VexRiscv.scala:123)
[error] 	at vexriscv.Pipeline$$anonfun$1.apply$mcV$sp(Pipeline.scala:161)
[error] 	at spinal.core.Component$$anonfun$prePop$1.apply(Component.scala:162)
[error] 	at spinal.core.Component$$anonfun$prePop$1.apply(Component.scala:160)
[error] 	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error] 	at spinal.core.Component.prePop(Component.scala:160)
[error] 	at spinal.core.Component.postInitCallback(Component.scala:169)
[error] 	at vexriscv.demo.Briey$$anon$3$$anon$4.<init>(Briey.scala:315)
[error] 	at vexriscv.demo.Briey$$anon$3.<init>(Briey.scala:310)
[error] 	at vexriscv.demo.Briey.<init>(Briey.scala:260)
[error] 	at vexriscv.demo.Briey_UzL$$anonfun$main$1.apply(Briey.scala:515)
[error] 	at vexriscv.demo.Briey_UzL$$anonfun$main$1.apply(Briey.scala:514)
[error] 	at spinal.core.internals.PhaseCreateComponent$$anonfun$impl$62.apply(Phase.scala:2350)
[error] 	at spinal.core.fiber.Engine$$anonfun$create$1.apply$mcV$sp(AsyncCtrl.scala:147)
[error] 	at spinal.core.fiber.AsyncThread$$anonfun$1.apply$mcV$sp(AsyncThread.scala:59)
[error] 	at spinal.core.fiber.EngineContext$$anonfun$newJvmThread$1.apply$mcV$sp(AsyncCtrl.scala:39)
[error] 	at spinal.sim.JvmThread.run(SimManager.scala:51)
[error] Nonzero exit code returned from runner: 1
```
The configuration with the DBusCachedPlugin and both IBusSimplePlugin and IBusCachedPlugin works properly.
Is this issue solvable (as you already mentioned that the integration with the Non-Cached Plugin currently probably is not working)?

Dolu1990, 2022-07-22:
So, the FPU integrate itself on the data bus via that trait : 
https://github.com/SpinalHDL/VexRiscv/blob/8d6cb2642104ced50f4259a66897894da51ca318/src/main/scala/vexriscv/plugin/DBusCachedPlugin.scala#L23

Only the DBusCachedPlugin implement it so far.
I guess it would be possible to implement it on the DBusSimplePlugin, but it may be quite a bit of work.

everetto89, 2022-07-25:
Great thanks a lot! I think this issue now can be closed.

closed 2022-07-25 by everetto89