SpinalHDL/VexRiscv #259: DataCache Supporting Multiple Cycle RAMs?

dockside-code, 2022-07-30:
Hi! I'm doing some research regarding different memory technologies and how they would interact with existing RISC-V architecture, and I'm trying to integrate a multiple-cycle RAM into the data cache to replace the existing data/array banks.

I've noticed that the datacache is pretty tightly coupled to the pipeline. But I'd like to know how feasible would this be? Is it possible without overhauling the entire existing pipeline? What are the main things/plugin logic that I would need to change? 

(I'm guessing probably the datacache and the cached databus, but anything else that I should probably need to keep an eye on? If i set the haltByOther signals properly would the pipeline still function?)

Thanks a lot!

Dolu1990, 2022-08-03:
Hi ^^

Hmm currently the data / instruction cache design should be configurable to push logic away up to provide 2 cycles read latency  : 

```
       address
         │
         ▼
     ┌─────────┐
─────┤         ├────────
     └─┬─────┬─┘
       │read │
       │     │
       └──┬──┘
          │
     ┌────▼────┐
─────┤         ├─────────
     └───┬─────┘
         │
         ▼
       Logic
```

For the writes, things are designed assuming that :
- cycle 1 : write ram[A] = B
- cycle 2 : read ram[A]
- cycle 3 : get result B

Read during write on the same address can be undefined behaviour.

Overall, getting more than two cycle latency ram read is quite tricky in Vex, and would need some redesign, idealy, adding more pipelining stages, either halting the pipeline. tricky changes to do ^^

So, else, in NaxRiscv, things are way more flexible in that regard, as pipeline can be stretched out, see : 
https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/fetch/FetchCachePlugin.scala#L181

dockside-code, 2022-08-03:
Thank you for the detailed response!
I've decided to stick with Vex., so good luck to me XD. 
I've been trying to revamp the data cache (with some success actually with the refilling process)
The problem I'm having now is to stall the pipeline when there's a read/write request to the cache until the data array inside the cache returns ready.
Could you please tell me a little bit more how `arbitration.haltByOthers` works , and what's a `Stageable()` ?

Dolu1990, 2022-08-03:
arbitration.haltByOther is to know in a given stage, when you are blocked because of stuff in futher stages.
"Stageable" is to define a hardware data type that you can inject in the pipeline and retreive in a futher stage. the instance of Stageable is kind of used as a "key" to retreive stuff from the pipeline ^^

dockside-code, 2022-08-03:
Thanks! I think I'll play with them for a little bit first.

closed 2022-08-03 by dockside-code

karegoud, 2024-08-09:
@dockside-code  I want to work on cache in vex can you give me some guidance on where should I start from and how to proceed. It will really help me a lot with my work.

dockside-code, 2024-08-09:
> @dockside-code I want to work on cache in vex can you give me some guidance on where should I start from and how to proceed. It will really help me a lot with my work.

Hi! So there’s actually a lot going on in the datacache file, a lot of features you may not be using! I would start by removing features you don’t need. Halting the pipeline is handled by the „haltIt” signals. Accessing the cache is handled by “data/tagReadCmd” and “data/tagWriteCmd” bundles. Hope this helps!

karegoud, 2024-08-09:
> Hi! So there’s actually a lot going on in the datacache file, a lot of features you may not be using! I would start by removing features you don’t need. Halting the pipeline is handled by the [...]

Thanks a lot for replying! So actually I want to work on creating a custom instruction for fft that requires lot of data which i want to access from cache after loading it from memory and also i want to have a part of cache only dedicated to my fft data. Is this possible in vex, if so can you guide me how to proceed

dockside-code, 2024-08-09:
> Thanks a lot for replying! So actually I want to work on creating a custom instruction for fft that requires lot of data which i want to access from cache after loading it from memory and also i [...]

Well I think it will be a lot less complicated that way. The way I’d do it is copy the file and make a new component called FFT_cache or smth, and reroute the upstream data. So the data upstream comes from smth like cpucachebus, use a stream demux (see spinalHDL doc) and route the request to fft_cache if address matches a certain region in which you store the FFT data.

karegoud, 2024-08-09:
> Well I think it will be a lot less complicated that way. The way I’d do it is copy the file and make a new component called FFT_cache or smth, and reroute the upstream data. So the data upstream [...]

Thanks. I will try the way you have suggested

dockside-code, 2024-08-16:
> Did you toke a look to VexiiRiscv ?

Awesome, is it linux capable yet?

Dolu1990, 2024-08-17:
Yes, it can also run debian / alpine
But note if you want more than 2 cycle latency block ram, then its L1 will not be great.
Better to look at the L2 ^^