SpinalHDL/VexRiscv #262: Illegal AMOADD instruction using IMAC variant in LiteX

boondockenergy, 2022-08-27:
I'm building a small RISCV Kernel in Rust on LiteX+VexRiscv, and came across an illegal instruction exception trying to use `Arc` (atomic reference counters) which is generating an `amoadd.w` instruction.

```
--============= Liftoff! ===============--
  _______
 |__   __|
    | | ___ _ __   ___  _ __
    | |/ _ \ '_ \ / _ \| '__|
    | |  __/ | | | (_) | |
    |_|\___|_| |_|\___/|_|
Rust RISCV Kernel Booting
[INFO] - Logging initialized
[INFO] - Heap start: 0x40125e04 end: 0x48000000 size: 126 MiB
[INFO] - Heap initialized
[INFO] - Interrupts enabled
[INFO] - Allocated stack 0x40125e04 (0x40125E04)
[INFO] - Thread entry point: 0x40001044
[INFO] - Allocated stack 0x40127e04 (0x40127E04)
[INFO] - Thread entry point: 0x40001014
[INFO] - EXCEPTION IllegalInstruction code=2
[INFO] -    MEPC: 0x40001CBE
[INFO] -   MTVAL: 0x00B5252F
[INFO] -  MCAUSE: 0x00000002
[INFO] - MHARTID: 0x00000000
```

Disassembly

```
40001cbe:       00b5252f                amoadd.w        a0,a1,(a0)
```

Does IMAC support AMO, or just LR/SC?

boondockenergy, 2022-08-27:
With https://github.com/litex-hub/pythondata-cpu-vexriscv/pull/19 I no longer get an illegal instruction.

```
  _______
 |__   __|
    | | ___ _ __   ___  _ __
    | |/ _ \ '_ \ / _ \| '__|
    | |  __/ | | | (_) | |
    |_|\___|_| |_|\___/|_|
Rust RISCV Kernel Booting
[INFO] - Logging initialized
[INFO] - Heap start: 0x40125e04 end: 0x48000000 size: 126 MiB
[INFO] - Heap initialized
[INFO] - Interrupts enabled
[INFO] - Allocated stack 0x40125e04 (0x40125E04)
[INFO] - Thread entry point: 0x40001044
[INFO] - Allocated stack 0x40127e04 (0x40127E04)
[INFO] - Thread entry point: 0x40001014
[INFO] - Adding Thread 1 [stack=0x40127e04 state=New]
[INFO] - TP: 0x40125bdc
[INFO] - Main thread started
[INFO] - Thread pointer 40129D64
[INFO] - Allocated stack 0x40129e70 (0x40129E70)
[INFO] - Thread entry point: 0x4000102c
[INFO] - Allocated stack 0x4012be70 (0x4012BE70)
[INFO] - Thread entry point: 0x4000107a
[INFO] - Adding Thread 3 [stack=0x4012be70 state=New]
[INFO] - Inner thread
```

Dolu1990, 2022-08-29:
Nice catch :)
Thanks !

Dolu1990, 2022-08-29:
(merged)

closed 2022-08-29 by Dolu1990