SpinalHDL/VexRiscv #263: [Question] How can I run C code on vexRiscv?

adz0612, 2022-08-31:
Hey, I have generated the Full version of vexRiscv using 
`sbt "runMain vexriscv.demo.GenFull" `

and then I ran some regression tests which passed.
I tried running the uart.elf example using openOCD and GDB,

`riscv32-unknown-elf-gdb src/test/resources/elf/uart.elf `
`load`
`continue`
This prints the output in the terminal window.

I want to understand how I can run any other C code following the same steps. I tried running a simple Hello world but did not get any output on the terminal window. I compiled the C code with
RV64 tool chain with -march="rv32ima" -mabi="ilp32" and then tried debugging the same like the uart example,
`riscv32-unknown-elf-gdb helloWorld.out `
`load`
`continue`

This did not print anything on the screen. I'm assuming I'm compiling the program wrong. Can someone help?

adz0612, 2022-09-01:
Hey, @Dolu1990 any idea how I can do this? Thanks!

tcal-x, 2022-09-01:
@adz0612 by any chance are you a user of LiteX?   It has a bare-metal demo that was my starting point for C programming on Vex.   There is probably the equivalent demo somewhere in the VexRiscv or Murax repos, but I don't know where.

Basically, you need a linker script to put the code and data regions in the correct spots for your system.

Dolu1990, 2022-09-02:
Here is a old example for Murax : 
https://github.com/SpinalHDL/VexRiscvSocSoftware/tree/master/projects/murax/demo

Very old stuff, not very nice XD

adz0612, 2022-10-20:
I was able to run C code using the Murax files, thanks!

closed 2022-10-20 by adz0612