SpinalHDL/VexRiscv #264: data cache stuck

hchoi-lattice, 2022-09-01:
Hi Dolu1990,
Found a weird D-cache stuck issue. Attached the waveform. It's related to refilling. D-cache goes into line fill phase and expect loader_counter_values to go up. On the other hand, dBus_cmd_valid is not asserted and it prevents main_cpu_dbus_cyc from being asserted. So, external memory controller does not know that it needs to send data to D-cache, while D-cache is waiting for the data. This makes a deadlock. 
Interestingly, the old version of VexRiscv (Generator : SpinalHDL v1.3.6    git head : 9bf01e7f360e003fac1dd5ca8b8f4bffec0e52b8) does not have this issue and works well.
Thanks,
Hoon

[image]

Dolu1990, 2022-09-02:
Hi ^^
Weird, would there be a way for me to get the wave file of the case ? That would realy help :D

hchoi-lattice, 2022-09-07:
Confirmed that the same issue was reported and already solved in Linefill Question #170. 
Thanks!

Dolu1990, 2022-09-07:
Hoo so with upstream version of VexRiscv, everything is good on your side ?