SpinalHDL/VexRiscv #265: AXI IO interface with VexRISCV SMP

goodwilj, 2022-09-09:
Hello, 

I'm looking to generate the Verilog/Vhdl of a VexRISCV multicore SMP processor but rework its IO buto AXI. I see it has been done for a single VexRISCV core:

https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/VexRiscvAxi4WithIntegratedJtag.scala

Is it possible to generate a multicore SMP VexRISCV with an AXI IO interface (which can then be packaged and used in a Vivado block design flow)?

Do I modify the IBUS and DBus plugins in the same manner in 
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/smp/VexRiscvSmpCluster.scala?

Regards,
Justin

P.S. very impressive work on VexRISCV and NaxRISCV :)

Dolu1990, 2022-09-12:
Hi ^^
The main interconnect for the soc has to stay in bmb, then when it goes to the main memory, you can add a bridge to AXI

So either you edit the scala, or you can the already generated Litex port SoC, and bridge its interface to AXI.
https://github.com/litex-hub/pythondata-cpu-vexriscv_smp/tree/master/pythondata_cpu_vexriscv_smp/verilog

Does it seems faisable ?

Regards
Charles

goodwilj, 2022-09-13:
Thanks for the quick reply! Yes, these are the two approaches I had considered.

1) For the scala approach, I see there is a bmb-to-AXI4 bridge, which I can use for a master AXI4 peripheral interface. However, for coherent DMA, I don't see an AXI4-to-bmb bridge. Since the DBusCachedPlugin is write-through, I think non-coherent DMA will be ok but may affect performance slightly. Is that correct? I only see coherent DMAs being used in the Litex SoC and the SaxonSoC.

2) The Litex SoC uses a bmb-to-wishbone bridge for the peripheral bus and wishbone-to-bmb bridge for a coherent dma. I could use the Wishbone to AXI4 bridge (which first converts to AXI4 Lite), but that seems a little convoluted going from bmb -> Wishbone -> AXI4-Lite -> AXI4, and the reverse for a coherent DMA. These bridges could add quite a bit of bloat in terms of resource utilization. Do you see any better approach for Litex option?

It seems the scala approach may be more elegant. What do you think?

Thanks,
Justin