SpinalHDL/VexRiscv #266: Documentation for 'simple' ibus/dbus protocol missing

Spritetm, 2022-09-25:
Seems Vexriscv by default has a 'simple' Ibus/Dbus implementation to fetch instructions and read/write data. What are the semantics of this bus? It seems AXI-like, but I cannot really find any details anywhere.

Spritetm, 2022-09-25:
Specifically, I can find data for the command stream at https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Libraries/stream.html, which is great. What I cannot find is requirements for the response; for instance am I allowed to 'batch accept' request and send the responses after they're all accepted? It seems like if I do that, at least with a write request followed by a read, I see issues.

Dolu1990, 2022-09-26:
Hi,

There is some doc about the native busses here : 
https://github.com/SpinalHDL/VexRiscv#ibussimpleplugin
https://github.com/SpinalHDL/VexRiscv#dbussimpleplugin

But it is missing for the cached busses. I just pushed some doc for the cached version.
https://github.com/SpinalHDL/VexRiscv/commit/7b9891829a3498c0178eaca660ab95cfc8f93390

Also note that there is a few bridges implemented, for instance : 
https://github.com/SpinalHDL/VexRiscv/blob/051d140c33ce1480e10bdf76668fceae8ff59bef/src/main/scala/vexriscv/ip/DataCache.scala#L317

For the stream yes, you are right for the doc, also it is kind of like AXI ^^

> I allowed to 'batch accept' request and send the responses after they're all accepted? It seems like if I do that, at least with a write request followed by a read, I see issues.

So, it is allowed to have multiple outstanding stuff (batch accept), as long as the slave do not impose it (meaning the slave has  to execute stuff soon or later)

One thing, write cmd should not generate read rsp for the l1 D$. Also VexRiscv need the memory bus to preserve ordering, so AXI4 is kind of a shity for that, as read/write channels aren't ordered.

Are you using the cached or uncached version of VexRiscv ?

There is already some AXI bridge implemented in the codebase, have you seen  : 
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/VexRiscvAxi4WithIntegratedJtag.scala ?