SpinalHDL/VexRiscv #267: FPU RegFile Details

gyh1997127, 2022-09-27:
Hi,

Does VexRiscv implement dedicated floating-point registers? If so, how does it implement double precision registers with a 32-bit base core? Thank you!

Dolu1990, 2022-09-27:
Hi,

> Does VexRiscv implement dedicated floating-point registers? 

Yes, via the FpuPlugin.

> If so, how does it implement double precision registers with a 32-bit base core? Thank you!

The main change in the core to handle it, is about supporting 64 bits load/store.
So to support it the data cache data path of VexRiscv is crancked up to 64 bits. Else nothing realy change, in the core, as the FpuPlugin does all the job of handeling the FPU register file and data path.

FPU register file : 
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/ip/fpu/FpuCore.scala#L158

Also note that internaly, floating point values are stored in a recoded format in order to avoid having to deal with subnormal.

gyh1997127, 2022-09-27:
Appreciate the insight!

closed 2022-09-27 by gyh1997127

gyh1997127, 2022-11-04:
Sorry about having to dig this one up, but what does the floating-point register file map to if `asyncRegFile == false`? 
I thought it would be BRAM, but the synthesis report says otherwise. Thank you for your insight.

Dolu1990, 2022-11-06:
> asyncRegFile == false? I thought it would be BRAM

https://github.com/SpinalHDL/VexRiscv/blob/051d140c33ce1480e10bdf76668fceae8ff59bef/src/main/scala/vexriscv/ip/fpu/FpuCore.scala#L285

So, mostly it specify the register file read in a clocked way if turn to false (to easy block ram inferation).
But still, the synthesis tool as the freedoom to emulate that syncronus read with distributed ram in order to save block ram usage. Especialy as there is only 32 words.