SpinalHDL/VexRiscv #268: VexRiscv not generating AR_VALID for READ

sarmadsalman-RS, 2022-10-04:
Hey there,
So i have been trying to use the VexRiscvAXI processor, with an open source AXI4 interconnect, to access an AXI RAM. I have made an example that can be looked up at the following repository: -
https://github.com/sarmadsalman-RS/VexRiscv_AXI_RAM

I have been able to write and read to an AXI RAM via VexRiscv, but i am unable to read more than once from the same address. The VexRiscv fails to generate the AR_VALID and as a result, the read does not happen. I can read once just fine from any memory aligned address, but if i try to read it again, the VexRiscv fails to assert the AR_VALID. All the required RTL, bare metal code along with a picture of a wavedump can be found at the following repository. A snippet of the wavedump can also be found below, and it is visible that the vexriscv generates the address but fails to assert the AR_VALID signal as shown: -
[image]

Continuous write operations work just fine, but reading more than once from same memory address fails. Any leads would be helpful.

Dolu1990, 2022-10-04:
Hi,

Which vexriscv configuration are you using ? with or without i$ d$ ?

If you check the cpu "lastStage****" signals in the CPU, they freez ?

sarmadsalman-RS, 2022-10-04:
@Dolu1990 in fact i have uploaded the wavedump at the following address: 
https://github.com/sarmadsalman-RS/VexRiscv_AXI_RAM/tree/master/VexRiscv_AXI_RAM

The address 00000010 has been read once in the beginning of this wavedump, and now at the position of the cursor, i am trying to again read the same address but it is not being read. The address is available on the dBus for a small time and in this time the AR_VALID fails to be asserted.

sarmadsalman-RS, 2022-10-05:
@Dolu1990 is there a pre-defined configuration for the VexRiscv CPU with AXI4_full interface without the cache plugins that we can just make and use or do we have to manually remove the cache plugins from the said configuration?

sarmadsalman-RS, 2022-10-05:
@Dolu1990 the program passes when using no-cache configuration, but it fails when run on the cache implemented CPU. Even when we update the data on a specific address, the second or more read instructions from that address fails because it had already read from that address once with the previous data. The DBusCache should be updated once the data on a specific address is updated and hence the new data should be read from the memory.

Dolu1990, 2022-10-05:
Hmm, can you send me the wave file ?

sarmadsalman-RS, 2022-10-05:
You can grab the wave file from the link below. You may also find the "main.c" file in the same repository to make sense of what is happening in the waveform. 

https://github.com/sarmadsalman-RS/VexRiscv_AXI_RAM/blob/master/VexRiscv_AXI_RAM/tb.vcd

sarmadsalman-RS, 2022-10-05:
When i modify the data on an address which i have read from earlier, i can see the response from the memory that the new data is written to the memory. If the write response comes valid for the new data, according to my understanding the cache should be refreshed simultaneously to give the new value upon receive instruction. 

Can you maybe point me from where i can see some description or an example on how to do cache invalidation via bare metal C where we have access to the addresses?

Dolu1990, 2022-10-05:
> according to my understanding the cache should be refreshed simultaneously to give the new value upon receive instruction.

Yes, when the CPU does a store, if the data cache has the corresponding cache line in cache, it  will update the cache line too.

To invalidate the data cache you can do :
asm(".word(0x500F)");

sarmadsalman-RS, 2022-11-02:
This works, thanks!

closed 2022-11-02 by sarmadsalman-RS