SpinalHDL/VexRiscv #269: Stuck in Subtraction function

ducdracaena, 2022-10-18:
Hi,

I tried to write the rust code as the firmware for vexriscv but the program is stuck at subtraction. 

Please help me to check it? Here is 2 files rust code and scala code.
https://file.io/IM591l9kBXZc

Thank you,
Duc

Dolu1990, 2022-10-18:
Hi, 
Can you send a wave file and the .elf ?

Thanks
Charles

ducdracaena, 2022-10-18:
Hi,

Here is the bin files https://www.file.io/mtpg/download/IikkfRY6JIbq

[image]
  Program nevers read, means never pass through subtraction

Thank you so much,
Duc

Dolu1990, 2022-10-18:
I mean, can you send a full wave file ? not a screen shot. 
I can for instance open a VCD file.

Thanks
Charles

ducdracaena, 2022-10-18:
Hi,

Can you please read this file? https://file.io/hFYpM9rO4Yxj

Thanks,
Duc

Dolu1990, 2022-10-18:
Hi, Yes i can open it :D

So, the CPU doesn't seems to freeze. The issue i have is that you sent the software .bin file, not the .elf
Do you have the executable (elf file) or assembly file (.asm) ?

ducdracaena, 2022-10-18:
Hello, 

I dont have the elf file, I sent you the rust code. I think you can convert that code to elf file? If not, I will ask my software engineer to do that. 

Thank you,
Duc

Dolu1990, 2022-10-18:
Hi,

> I think you can convert that code to elf file?

No it can't be done, as all symboles are lost in the bin file.

>  I sent you the rust code

The issue with that method is that  if i try to compile it myself i will get a different binary, as compiler version and stuff like that will come to play.

Thanks ^^
Charles

ducdracaena, 2022-10-19:
Hi,

I am trying to create the elf file. 

But on your side, with subtraction, the cpu is still working?

Thanks ^^
Duc

ducdracaena, 2022-10-19:
Hi, 

Here is elf file :)
https://file.io/69LnFEirq9OD

Thank you,
Duc

Dolu1990, 2022-10-19:
> But on your side, with subtraction, the cpu is still working?

I'm not trying to reproduce the issue yet, as it can easily lead nowhere, first i realy need to see the issue with your setup.
That's why i asked for the wave file (vcd) and the .asm xor .elf files in order to look at the wave and figure out what the CPU is doing by cross checking the .asm/.elf file.

The https://file.io/69LnFEirq9OD file seem to be deleted after i downloaded it once ? (sim_debug.example right ?)
Seems it is just a diff file, not a elf nor a asm file, right ?

ducdracaena, 2022-10-19:
Hi, 

it is the elf file. I uploaded again here https://file.io/vcHhNfqf4tL9

Thank you
Duc

Dolu1990, 2022-10-19:
Hi,

Looking at the assembly code, i can see there is some LD and SD instruction which are only for RV64. Seems like the firmware you sent me wasn't compiled for RV32 ?

ducdracaena, 2022-10-20:
Hi,

It seems when I tried to complie for naxriscv, it is RV64 so I need to change to RV32 for vexriscv :) Let me try again. 

Thank you so much, 
Duc ^^

ducdracaena, 2022-10-20:
Hi, 

Issue is solved :) I used RV32 for vexriscv

Thank you so much,
Duc

closed 2022-10-20 by Dolu1990