SpinalHDL/VexRiscv #270: How to access data from memory location in Custom instruction?

adz0612, 2022-10-20:
Hey, I want to access data from a particular address in my custom instruction. This address will be supplied to the instruction as input. Is there a way I can do this?

Something like:
custInst(r1,r2,rd)

rd = data[r1] + data[r2] 

Also, can I access the temporary registers (Like x29, x30 etc.) inside the custom instruction?

Dolu1990, 2022-10-20:
Hi,

> access data from a particular address

From the main memory using the data cache ?
To sum two main memory words ?
It isn't possible, i mean, the data cache pipeline is made to serve only one thing at the time.

> Also, can I access the temporary registers (Like x29, x30 etc.) inside the custom instruction?

Currently, only if they are provided via RS1 / RS2 field, following the RISC-V instruction encoding.

adz0612, 2022-10-20:
More specifically, I am trying to access data stored in an array. I wanted to send the base address of the array as the input to the custom instruction and get all the elements of the array by incrementing the index internally.

I'd want to do that for multiple arrays.
something like: 

`custInst(baseAddrOfArray1, baseAddrOfArr2, rd)`

Is something like that possible? From your answer, I understand there isn't a way to do this.

Dolu1990, 2022-10-20:
Not possible with the current design :)

adz0612, 2022-10-22:
Ahh alright, thanks!

closed 2022-10-22 by adz0612