SpinalHDL/VexRiscv #277: How to use custom instruction in C code?

adz0612, 2022-11-10:
I have a custom instruction implemented say, `CustomMac` How do I go about using this in a C code?
If my instruction does:

**CustomMac(rd,rd1,rd2)**
rd = rd1+rd2 ;

How do I translate to:

**C Code:**

int main(){
int a,b,c;
b=5;
c=5;

CustomMac(a,b,c);
// a should have 10 as the result
}

Dolu1990, 2022-11-10:
Hi, 
Here is an example : 
https://spinalhdl.github.io/NaxRiscv-Rtd/main/NaxRiscv/execution_units/index.html#software-test

adz0612, 2022-11-10:
Hi, @Dolu1990 

I'm specifically looking for inline assembly usage. I have seen the crt.S file in your SIMD_ADD example and I was trying something similar to 

asm volatile ("SIMD_ADD(a,b,c)")
How can I make something like this work?

Dolu1990, 2022-11-10:
Hooo my bad.
For C inline assembly, see that example  : 

https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.3/software/standalone/driver/aes_custom.h#L5

tcal-x, 2022-11-10:
I found the macros here are useful: https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.1/software/standalone/driver/riscv.h 

My use of those macros then looks like this:
```
#define cfu_op_hw(funct3, funct7, rs1, rs2)    opcode_R(CUSTOM0, funct3, funct7, (rs1), (rs2))
```

In C code the use looks like
```
unsigned result = cfu_op_hw(0, 0, x, y);
```
where result, x, and y are just C variables.

Of course you can then define your own macros to make it look like
```
unsigned result = MY_OP(x, y);
```

adz0612, 2022-11-11:
Thanks a lot for the suggestions, I did something similar to what @tcal-x suggested which is also done at #206 and in Dolus answer. 

I was able to get it working. Thanks!

closed 2022-11-11 by adz0612

karegoud, 2024-09-09:
@adz0612  Can you please tell me in which file we have to write the inline assembly to use my custom instruction for doing further computations and how to verify it.

ashuthosh-mr, 2024-09-18:
In the main C file

karegoud, 2024-09-19:
@ashuthosh-mr  Thanks for replying. I wrote inline in main.c file and also added my plugin in Murax.scala file in demo folder but I am unable to get output. I also don't see my plugin added to cpu in test/cpp/murax/Murax.fst file.
Are there anymore changes that I am missing.

karegoud, 2024-09-25:
@Dolu1990 Can you help me with the above question I asked

ashuthosh-mr, 2024-09-26:
Can you share the C code skeleton that you are using to test the custom plugin.

karegoud, 2024-09-29:
#include "riscv.h" 
#define fft_round(rs1, rs2, func3, func7) opcode_R(CUSTOM0, func3, func7, rs1, rs2)
static inline __attribute__((always_inline)) void vexriscv_fft_computation (unsigned int *in,
                          unsigned int *out)
                         {
   unsigned int  s[2];   
   unsigned int  t[2];        
   
        s[0] = ((volatile unsigned int*)in)[0];
        s[1] = ((volatile unsigned int*)in)[1];
        
        t[0] =  fft_round(s[0], s[1], 0x00, 0);
        t[1] = fft_round(s[0], s[1], 0x00, 64);
        
        ((volatile unsigned int*)out)[0] = t[0];
        ((volatile unsigned int*)out)[1] = t[1];
        
        }
        
      void main() {
    // Example input data (plaintext) - 16 bytes for AES-128
    unsigned int input[2] ; 
    input[0] = 0x00000003  ;
    input[1] = 0x00000003 ;

    unsigned int output[2];

    // Perform AES encryption
    vexriscv_fft_computation(input, output);
   //println("output of fft computation is: ");
 
}

How to debug this if I am getting an error. In Murax.fst file I am not getting any signals related to this custom instruction even though  I used TRACE=yes in my commands. This custom instruction was working fine when verified with assembly language [using crt.S file
]

ashuthosh-mr, 2024-10-04:
Not sure of the above code. Try this:

`asm(".set abiname_zero, 0");
asm(".set abiname_ra , 1");
asm(".set abiname_sp , 2");
asm(".set abiname_gp , 3");
asm(".set abiname_tp , 4");
asm(".set abiname_t0 , 5");
asm(".set abiname_t1 , 6");
asm(".set abiname_t2 , 7");
asm(".set abiname_s0 , 8");
asm(".set abiname_s1 , 9");
asm(".set abiname_a0 , 10");
asm(".set abiname_a1 , 11");
asm(".set abiname_a2 , 12");
asm(".set abiname_a3 , 13");
asm(".set abiname_a4 , 14");
asm(".set abiname_a5 , 15");
asm(".set abiname_a6 , 16");
asm(".set abiname_a7 , 17");
asm(".set abiname_s2 , 18");
asm(".set abiname_s3 , 19");
asm(".set abiname_s4 , 20");
asm(".set abiname_s5 , 21");
asm(".set abiname_s6 , 22");
asm(".set abiname_s7 , 23");
asm(".set abiname_s8 , 24");
asm(".set abiname_s9 , 25");
asm(".set abiname_s10 , 26");
asm(".set abiname_s11 , 27");
asm(".set abiname_t3 , 28");
asm(".set abiname_t4 , 29");
asm(".set abiname_t5 , 30");
asm(".set abiname_t6 , 31");

///////////
#define SIMD_ADD(src1, src2, result) \
asm volatile(  \
".word ((0b0000011 << 25) | (abiname_%2 << 20) | (abiname_%1 << 15) | (0b000 << 12) | (abiname_%0 << 7) | 0b0110011) \n\t" \
: "=r"(result)    \
: "r" (src1), "r" (src2) \
) 
`
Place the above piece in the main.c, you can call SIMD_ADD( a,b,c); in the void main() function. 

Let me know if you need the definition to include the above plugin in Murax configuration.

NOTE: Although it is called SIMD_ADD, the functionality in the plugin is changed to desired functionality.

karegoud, 2024-10-05:
May I get the definition to include plugin in Murax configuration

ashuthosh-mr, 2024-11-20:
You can include in the cpu plugins of the file Murax.scala inside demo folder. Add the SimdAdd plugin. You can refer to GenwithSimdaddplugin.scala. Hope that helps.