SpinalHDL/VexRiscv #278: Error during Synthesis of Murax VHDL with Synplify Pro

robindust-ce, 2022-11-10:
I ran into some weird problems when synthesising SpinalHDL generated VHDL with Synplify Pro (Distributed as part of Lattice Diamond Version 3.12.1.454) and was wondering if anyone else ran into that. Specifically, Synplify in some cases fails to evaluate the widths of "pkg_[...]" functions in the VHDL Code of the Murax SoC (default config). 

These are the examples that I encountered:

Line 6673 (and the following two) in the Apb3Decoder:

`io_output_PSEL_read_buffer(0) <= (pkg_toStdLogic((io_input_PADDR and pkg_not(pkg_unsigned("00000000111111111111"))) = pkg_unsigned("00000000000000000000")) and pkg_extract(io_input_PSEL,0));`

Which generates this error:

> ERROR - CD150 :"D:\projects\murax_rl_xo3l\Murax.vhd":7363:94:7363:115|Width mismatch, variable value has width 1, value 20

And line 5238 in the JtagBridge:

`jtag_tap_instructionShift <= pkg_resize(pkg_stdLogicVector("01"),4);`

Which generates this error:

> ERROR - CD150 :"D:\projects\murax_rl_xo3l\Murax.vhd":5160:69:5160:72|Width mismatch, variable that has width 1, value 2

I can remove the errors by replacing the "pkg_not" with the VHDL native "not" and the "pkg_resize" by a literal ("0001"). The use of the pkg_[...] functions in the rest of the code is synthesized without errors. I don't see a pattern why those lines cause errors but others don't.

I am assuming this must be a bug on Synplify side but was wondering if this is a known problem and whether there is any fix I can apply from the SpinalHDL side other than manually fixing the VHDL code?

Dolu1990, 2022-11-10:
Hi,

2 weeks ago, somebody else contacted me by email about the same issue.
And yes, seems like it is a symplify pro bug which kind of inline stuff while it shouldn't.

So i'm not sure if we there is a nice workaround. should use `std_logic_vector("01") but the funny thing is that it trigger a GHDL bug (which was fixed 1 week ago).

Hardware => riding between the tools bugs <3
Else, could you use the verilog versin of Murax ? That one should be fine with the tool i think.

robindust-ce, 2022-11-10:
Hi, thank you for the quick answer! I just wanted to confirm that I don't make some silly mistake causing this. I guess I will just stick to manually editing the VHDL or substitute it for Verilog as you suggested.

closed 2022-11-10 by robindust-ce