SpinalHDL/VexRiscv #281: All-Zeros Instruction

hqd8888, 2022-11-17:
Hello, when I try using all zeros instruction to go into trap, I find vex seems not to support it in compressed cores. It will convert the instruction to c.addi4spn but miss to check if imm is 0. I wonder if I could add a when statement to make all zeros instruction as illegal instruction with no side effects.

Dolu1990, 2022-11-17:
Hi,

I think this should fix it : 
https://github.com/SpinalHDL/VexRiscv/commit/36c3346e51a6d9b19acb12e0aaab4632f8822ff4

let's me know how it goes ^^

hqd8888, 2022-11-17:
> Hi,
> 
> I think this should fix it : [36c3346](https://github.com/SpinalHDL/VexRiscv/commit/36c3346e51a6d9b19acb12e0aaab4632f8822ff4)
> 
> let's me know how it goes ^^

Actually I made similar fix on my side and I am sure it works. I just want to confirm such modification will not affect other sections. ^^

Dolu1990, 2022-11-17:
Ahhh nice, cool :)

closed 2022-11-17 by Dolu1990