SpinalHDL/VexRiscv #282: object VJtagBridge is not a member of package spinal.lib.system.debugger

quantrpeter, 2022-11-17:
hi, i can't compile it, please help
[image]

```
java -version
openjdk version "1.8.0_352"
OpenJDK Runtime Environment (build 1.8.0_352-8u352-ga-1~22.04-b08)
OpenJDK 64-Bit Server VM (build 25.352-b08, mixed mode)
```

LongStudy, 2022-11-18:
I'm running this project for the first time, and the same error appear in ubuntu 20 and ubuntu 18

[error] /mnt/d/VexRiscv/src/main/scala/vexriscv/plugin/DebugPlugin.scala:4:8: object VJtagBridge is not a member of package spinal.lib.system.debugger
[error] import spinal.lib.system.debugger.{JtagBridge, JtagBridgeNoTap, VJtagBridge, SystemDebugger, SystemDebuggerConfig, SystemDebuggerMemBus}
[error]        ^
[error] /mnt/d/VexRiscv/src/main/scala/vexriscv/plugin/DebugPlugin.scala:175:26: not found: type VJtagBridge
[error]     val jtagBridge = new VJtagBridge(jtagConfig)
[error]                          ^
[error] two errors found
[error] (Compile / compileIncremental) Compilation failed
[error] Total time: 36 s, completed Nov 18, 2022 10:45:20 AM

siluks-plumerai, 2022-11-18:
workarund, just remove this `VJtagBridge` from `DebugPlugin.scala`
[image]

Dolu1990, 2022-11-18:
Hi,

It was due to a PR merged in masted instead of dev. the situation is now fixed :)

closed 2022-11-18 by Dolu1990