SpinalHDL/VexRiscv #284: How to reset default soc clock frequency on FPGA？

JuzhongZheng666, 2022-12-01:
Hi，I want to set the soc clock frequency, when I load the bitstream to the FPGA. How can I do it?
I try to add 'soc_kwargs' syntex as fllow:
[image]

But I meet this error:
[image]
[image]

Dolu1990, 2022-12-01:
Hi,

I'm not sure myself about how to do it the right way with Litex.
Looking at def __init__(self, sys_clk_freq=125e6, with_led_chaser=True, with_pcie=False, **kwargs): from the litex-board, i would say you can : 
Board.__init__(self, sys_clk_freq= ???, ...) ?

(overall the VexRiscv isn't the right repo to ask the question XD)

JuzhongZheng666, 2022-12-01:
Thank you for your answer very much. I will try. Sorry for ask this question here, I guess I make a mistake.

closed 2022-12-01 by JuzhongZheng666