SpinalHDL/VexRiscv #287: verilator 5 - simulation failed for Linux.scala

laurentiuduca, 2022-12-10:
hello

tried to simulate vexriscv
followed the instructions from Linux.scala

the linux files are
VexRiscvRegressionData/sim/linux

laur@laurPC-100:~/lucru/cn/riscv/vexriscv-linux/VexRiscv/src/test/cpp/regression$ verilator --version
Verilator 5.002 2022-10-29 rev v5.002-29-gdb39d70c7

export BUILDROOT=/home/laur/lucru/cn/riscv/vexriscv-linux/buildroot
laur@laurPC-100:~/lucru/cn/riscv/vexriscv-linux/VexRiscv/src/test/cpp/regression$ make clean run IBUS=CACHED DBUS=CACHED  DEBUG_PLUGIN=STD SUPERVISOR=yes CSR=yes DEBUG_PLUGIN=no  COMPRESSED=no LRSC=yes AMO=yes REDO=0 DHRYSTONE=no LINUX_SOC=yes EMULATOR=~/lucru/cn/riscv/vexriscv-linux/VexRiscvRegressionData/sim/linux/emulator/emulator.bin VMLINUX=$BUILDROOT/output/images/Image DTB=$BUILDROOT/board/spinal/vexriscv_sim/rv32.dtb RAMDISK=$BUILDROOT/output/images/rootfs.cpio WITH_USER_IO=yes TRACE=no FLOW_INFO=no
rm -rf obj_dir
rm -f VexRiscv.v*.bin
rm -f VexRiscv.v*.bin
cp ../../../../VexRiscv.v*.bin . | true
verilator -cc  ../../../../VexRiscv.v  -O3 -CFLAGS -std=c++11 -LDFLAGS -pthread  -CFLAGS -DIBUS_CACHED -CFLAGS -DDBUS_CACHED -CFLAGS -DREDO=0 -CFLAGS -pthread -CFLAGS -DTHREAD_COUNT=4 -CFLAGS -O3  -O3 -CFLAGS -DLINUX_SOC -CFLAGS -DVMLINUX='\"/home/laur/lucru/cn/riscv/vexriscv-linux/buildroot/output/images/Image\"' -CFLAGS -DDTB='\"/home/laur/lucru/cn/riscv/vexriscv-linux/buildroot/board/spinal/vexriscv_sim/rv32.dtb\"' -CFLAGS -DRAMDISK='\"/home/laur/lucru/cn/riscv/vexriscv-linux/buildroot/output/images/rootfs.cpio\"' -CFLAGS -DEMULATOR='\"/home/laur/lucru/cn/riscv/vexriscv-linux/VexRiscvRegressionData/sim/linux/emulator/emulator.bin\"' -CFLAGS -DTIMER_INTERRUPT -CFLAGS -DEXTERNAL_INTERRUPT -CFLAGS -DWITH_USER_IO=yes -CFLAGS -DSUPERVISOR -CFLAGS -DSTALL=1 -CFLAGS -DCSR -CFLAGS -DLRSC -CFLAGS -DAMO -CFLAGS -DISA_TEST -CFLAGS -DMMU -CFLAGS -DMUL -CFLAGS -DDIV -CFLAGS -DTRACE_START=0 --gdbbt  -Wno-UNOPTFLAT -Wno-WIDTH --x-assign unique --exe main.cpp
No stack.
make  -j4 -C obj_dir/ -f VVexRiscv.mk VVexRiscv
make[1]: Entering directory '/home/laur/lucru/cn/riscv/vexriscv-linux/VexRiscv/src/test/cpp/regression/obj_dir'
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -DIBUS_CACHED -DDBUS_CACHED -DREDO=0 -pthread -DTHREAD_COUNT=4 -O3 -DLINUX_SOC -DVMLINUX=\"/home/laur/lucru/cn/riscv/vexriscv-linux/buildroot/output/images/Image\" -DDTB=\"/home/laur/lucru/cn/riscv/vexriscv-linux/buildroot/board/spinal/vexriscv_sim/rv32.dtb\" -DRAMDISK=\"/home/laur/lucru/cn/riscv/vexriscv-linux/buildroot/output/images/rootfs.cpio\" -DEMULATOR=\"/home/laur/lucru/cn/riscv/vexriscv-linux/VexRiscvRegressionData/sim/linux/emulator/emulator.bin\" -DTIMER_INTERRUPT -DEXTERNAL_INTERRUPT -DWITH_USER_IO=yes -DSUPERVISOR -DSTALL=1 -DCSR -DLRSC -DAMO -DISA_TEST -DMMU -DMUL -DDIV -DTRACE_START=0  -std=gnu++14 -Os -c -o main.o ../main.cpp
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -DIBUS_CACHED -DDBUS_CACHED -DREDO=0 -pthread -DTHREAD_COUNT=4 -O3 -DLINUX_SOC -DVMLINUX=\"/home/laur/lucru/cn/riscv/vexriscv-linux/buildroot/output/images/Image\" -DDTB=\"/home/laur/lucru/cn/riscv/vexriscv-linux/buildroot/board/spinal/vexriscv_sim/rv32.dtb\" -DRAMDISK=\"/home/laur/lucru/cn/riscv/vexriscv-linux/buildroot/output/images/rootfs.cpio\" -DEMULATOR=\"/home/laur/lucru/cn/riscv/vexriscv-linux/VexRiscvRegressionData/sim/linux/emulator/emulator.bin\" -DTIMER_INTERRUPT -DEXTERNAL_INTERRUPT -DWITH_USER_IO=yes -DSUPERVISOR -DSTALL=1 -DCSR -DLRSC -DAMO -DISA_TEST -DMMU -DMUL -DDIV -DTRACE_START=0  -std=gnu++14 -Os -c -o verilated.o /usr/local/share/verilator/include/verilated.cpp
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -DIBUS_CACHED -DDBUS_CACHED -DREDO=0 -pthread -DTHREAD_COUNT=4 -O3 -DLINUX_SOC -DVMLINUX=\"/home/laur/lucru/cn/riscv/vexriscv-linux/buildroot/output/images/Image\" -DDTB=\"/home/laur/lucru/cn/riscv/vexriscv-linux/buildroot/board/spinal/vexriscv_sim/rv32.dtb\" -DRAMDISK=\"/home/laur/lucru/cn/riscv/vexriscv-linux/buildroot/output/images/rootfs.cpio\" -DEMULATOR=\"/home/laur/lucru/cn/riscv/vexriscv-linux/VexRiscvRegressionData/sim/linux/emulator/emulator.bin\" -DTIMER_INTERRUPT -DEXTERNAL_INTERRUPT -DWITH_USER_IO=yes -DSUPERVISOR -DSTALL=1 -DCSR -DLRSC -DAMO -DISA_TEST -DMMU -DMUL -DDIV -DTRACE_START=0  -std=gnu++14 -Os -c -o verilated_dpi.o /usr/local/share/verilator/include/verilated_dpi.cpp
/usr/bin/perl /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VVexRiscv.cpp VVexRiscv___024root__DepSet_hcd8610ab__0.cpp VVexRiscv___024root__DepSet_h0c12cad5__0.cpp VVexRiscv_VexRiscv__DepSet_hd894e8a5__0.cpp VVexRiscv_VexRiscv__DepSet_hd4e48a5e__0.cpp VVexRiscv__Dpi.cpp VVexRiscv__ConstPool_0.cpp VVexRiscv___024root__Slow.cpp VVexRiscv___024root__DepSet_hcd8610ab__0__Slow.cpp VVexRiscv___024root__DepSet_h0c12cad5__0__Slow.cpp VVexRiscv_VexRiscv__Slow.cpp VVexRiscv_VexRiscv__DepSet_hd894e8a5__0__Slow.cpp VVexRiscv_VexRiscv__DepSet_hd4e48a5e__0__Slow.cpp VVexRiscv__Syms.cpp > VVexRiscv__ALL.cpp
echo "" > VVexRiscv__ALL.verilator_deplist.tmp
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -DIBUS_CACHED -DDBUS_CACHED -DREDO=0 -pthread -DTHREAD_COUNT=4 -O3 -DLINUX_SOC -DVMLINUX=\"/home/laur/lucru/cn/riscv/vexriscv-linux/buildroot/output/images/Image\" -DDTB=\"/home/laur/lucru/cn/riscv/vexriscv-linux/buildroot/board/spinal/vexriscv_sim/rv32.dtb\" -DRAMDISK=\"/home/laur/lucru/cn/riscv/vexriscv-linux/buildroot/output/images/rootfs.cpio\" -DEMULATOR=\"/home/laur/lucru/cn/riscv/vexriscv-linux/VexRiscvRegressionData/sim/linux/emulator/emulator.bin\" -DTIMER_INTERRUPT -DEXTERNAL_INTERRUPT -DWITH_USER_IO=yes -DSUPERVISOR -DSTALL=1 -DCSR -DLRSC -DAMO -DISA_TEST -DMMU -DMUL -DDIV -DTRACE_START=0  -std=gnu++14 -Os -c -o VVexRiscv__ALL.o VVexRiscv__ALL.cpp
Archive ar -rcs VVexRiscv__ALL.a VVexRiscv__ALL.o
../main.cpp: In member function ‘virtual void Workspace::dBusAccess(uint32_t, bool, uint32_t, uint32_t, uint32_t*, bool*)’:
../main.cpp:1336:27: error: too many arguments to function ‘IData VL_RANDOM_I()’
 1336 |     *data = VL_RANDOM_I(32);
      |                           ^
In file included from /usr/local/share/verilator/include/verilated.h:947,
                 from ./VVexRiscv.h:11,
                 from ../main.cpp:1:
/usr/local/share/verilator/include/verilated_funcs.h:87:14: note: declared here
   87 | inline IData VL_RANDOM_I() VL_MT_SAFE { return vl_rand64(); }
      |              ^~~~~~~~~~~
../main.cpp: At global scope:
../main.cpp:1772:17: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
 1772 |  char *target = "PROJECT EXECUTION SUCCESSFUL", *hit = target;
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../main.cpp: In member function ‘virtual void ZephyrRegression::dutPutChar(char)’:
../main.cpp:1781:20: warning: NULL used in arithmetic [-Wpointer-arith]
 1781 |         if(*hit == NULL) {
      |                    ^~~~
../main.cpp: In member function ‘virtual void IBusCached::postCycle()’:
../main.cpp:1958:56: error: too many arguments to function ‘IData VL_RANDOM_I()’
 1958 |   if(pendingCount != 0 && (!ws->iStall || VL_RANDOM_I(7) < 100)){
      |                                                        ^
In file included from /usr/local/share/verilator/include/verilated.h:947,
[... cut ...]
../main.cpp:2820:11: warning: ignoring return value of ‘size_t fread(void*, size_t, size_t, FILE*)’, declared with attribute warn_unused_result [-Wunused-result]
 2820 |      fread(log, 1, logSize, logFile);
      |      ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
make[1]: *** [VVexRiscv.mk:87: main.o] Error 1
rm VVexRiscv__ALL.verilator_deplist.tmp
make[1]: Leaving directory '/home/laur/lucru/cn/riscv/vexriscv-linux/VexRiscv/src/test/cpp/regression/obj_dir'
make: *** [makefile:272: compile] Error 2
laur@laurPC-100:~/lucru/cn/riscv/vexriscv-linux/VexRiscv/src/test/cpp/regression$ verilator --version
Verilator 5.002 2022-10-29 rev v5.002-29-gdb39d70c7

laurentiuduca, 2022-12-10:
forgot to say that in verilator 4 it works

Dolu1990, 2022-12-12:
Hi,

You may have a old version of VexRiscv ? seems this has been fixed with https://github.com/SpinalHDL/VexRiscv/blob/master/src/test/cpp/regression/main.cpp#L23

laurentiuduca, 2022-12-12:
you are right
thank you

siatzjs, 2023-08-09:
Hi,I have encountered some issues,please reply at your convenience.
TART TEST test_id_99_IBus_Simple4S2PInjStagevexriscv.plugin.DYNAMIC_DBus_Cached64d32cS8192W2BPL64Rsp Amo Tsmmu Atm_MulDiv_MulDivFpgaSimple_Shift_FullEarly_Branch_Early_Hazard_BypassMemory_RegFile_AsyncDR_Src_AddSub_Csr_MachineOs_Decoder__Debug_None_DBus_NoMemProtect
[Runtime] SpinalHDL v1.9.0    git head : 7d30dbacbd3aa1be42fb2a3d4da5675703aae2ae
[Runtime] JVM max memory : 2425.0MiB
[Runtime] Current date : 2023.08.09 16:10:16
[Progress] at 15.564 : Elaborate components
[Warning] This VexRiscv configuration is set without software ebreak instruction support. Some software may rely on it (ex: Rust). (This isn't related to JTAG ebreak)
[Progress] at 15.703 : Checks and transforms
[Progress] at 15.796 : Generate Verilog
[Warning] toplevel/IBusSimplePlugin_rspJoin_rspBuffer_c/fifo/logic_ram : Mem[5*33 bits].readAsync can only be write first into Verilog
[Warning] toplevel/dataCache_1/ways_0_tags : Mem[64*22 bits].readAsync can only be write first into Verilog
[Warning] toplevel/dataCache_1/ways_1_tags : Mem[64*22 bits].readAsync can only be write first into Verilog
[Warning] toplevel/RegFilePlugin_regFile : Mem[32*32 bits].readAsync can only be write first into Verilog
[Warning] toplevel/RegFilePlugin_regFile : Mem[32*32 bits].readAsync can only be write first into Verilog
[Warning] 109 signals were pruned. You can call printPruned on the backend report to get more informations.
[Done] at 15.838
make run REGRESSION_PATH=../../src/test/cpp/regression VEXRISCV_FILE=VexRiscv.v WITH_USER_IO=no REDO=10 TRACE=yes TRACE_START=100000000000ll FLOW_INFO=no STOP_ON_ERROR=no DHRYSTONE=yes COREMARK=yes THREAD_COUNT=1  SEED=1526284434 IBUS=SIMPLE DBUS=CACHED DBUS_LOAD_DATA_WIDTH=64 DBUS_STORE_DATA_WIDTH=32 LRSC=yes AMO=yes  MUL=yes DIV=yes      CSR=yes CSR_SKIP_TEST=yes FREERTOS=0 ZEPHYR=4  DEBUG_PLUGIN=no MMU=no PMP=no
cp VexRiscv.v*.bin . | true
verilator -cc  VexRiscv.v  -O3 -CFLAGS -std=c++11 -LDFLAGS -pthread  -CFLAGS -DREGRESSION_PATH='\"../../src/test/cpp/regression/\"' -CFLAGS -DIBUS_SIMPLE -CFLAGS -DIBUS_DATA_WIDTH=32 -CFLAGS -DDBUS_LOAD_DATA_WIDTH=64 -CFLAGS -DDBUS_STORE_DATA_WIDTH=32 -CFLAGS -DDBUS_CACHED -CFLAGS -DREDO=10 -CFLAGS -pthread -CFLAGS -Wno-unused-result -CFLAGS -DTHREAD_COUNT=1 -CFLAGS -O3  -O3 -CFLAGS -DCOREMARK -CFLAGS -DWITH_RISCV_REF -CFLAGS -DTIMER_INTERRUPT -CFLAGS -DEXTERNAL_INTERRUPT -CFLAGS -DDHRYSTONE -CFLAGS -DSTALL=1 -CFLAGS -DSEED=1526284434 -CFLAGS -DTRACE -CFLAGS -DCSR -CFLAGS -DCSR_SKIP_TEST -CFLAGS -DLRSC -CFLAGS -DAMO -CFLAGS -DISA_TEST -CFLAGS -DMUL -CFLAGS -DDIV -CFLAGS -DTRACE_START=100000000000ll -CFLAGS -DFREERTOS -CFLAGS -DFREERTOS_COUNT=0 -CFLAGS -DZEPHYR -CFLAGS -DZEPHYR_COUNT=4 --gdbbt --trace-fst -Wno-UNOPTFLAT -Wno-WIDTH --x-assign unique --exe main.cpp
cp: 'VexRiscv.v_toplevel_RegFilePlugin_regFile.bin' and './VexRiscv.v_toplevel_RegFilePlugin_regFile.bin' are the same file
No stack.
make  -j1 -C obj_dir/ -f VVexRiscv.mk VVexRiscv
make[1]: Entering directory '/home/vboxuser/test/VexRiscv/simWorkspace/test_id_99_/obj_dir'
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -DREGRESSION_PATH=\"../../src/test/cpp/regression/\" -DIBUS_SIMPLE -DIBUS_DATA_WIDTH=32 -DDBUS_LOAD_DATA_WIDTH=64 -DDBUS_STORE_DATA_WIDTH=32 -DDBUS_CACHED -DREDO=10 -pthread -Wno-unused-result -DTHREAD_COUNT=1 -O3 -DCOREMARK -DWITH_RISCV_REF -DTIMER_INTERRUPT -DEXTERNAL_INTERRUPT -DDHRYSTONE -DSTALL=1 -DSEED=1526284434 -DTRACE -DCSR -DCSR_SKIP_TEST -DLRSC -DAMO -DISA_TEST -DMUL -DDIV -DTRACE_START=100000000000ll -DFREERTOS -DFREERTOS_COUNT=0 -DZEPHYR -DZEPHYR_COUNT=4  -std=gnu++14 -Os -c -o main.o ../main.cpp
../main.cpp: In member function ￢ﾀﾘvirtual void IBusSimple::postCycle()￢ﾀﾙ:
../main.cpp:23:42: error: too few arguments to function ￢ﾀﾘIData VL_RANDOM_I(int)￢ﾀﾙ
   23 | #define VL_RANDOM_I_WIDTH(w) (VL_RANDOM_I() & (1l << w)-1l)
      |                               ~~~~~~~~~~~^~
../main.cpp:2079:52: note: in expansion of macro ￢ﾀﾘVL_RANDOM_I_WIDTH￢ﾀﾙ
 2079 |                 if(rPtr != wPtr && (!ws->iStall || VL_RANDOM_I_WIDTH(7) < 100)){
      |                                                    ^~~~~~~~~~~~~~~~~
In file included from /usr/local/share/verilator/include/verilated_heavy.h:27,
                 from ./VVexRiscv.h:11,
                 from ../main.cpp:1:
/usr/local/share/verilator/include/verilated.h:935:14: note: declared here
  935 | inline IData VL_RANDOM_I(int obits) VL_MT_SAFE { return vl_rand64() & VL_MASK_I(obits); }
      |              ^~~~~~~~~~~
../main.cpp:23:42: error: too few arguments to function ￢ﾀﾘIData VL_RANDOM_I(int)￢ﾀﾙ
   23 | #define VL_RANDOM_I_WIDTH(w) (VL_RANDOM_I() & (1l << w)-1l)
      |                               ~~~~~~~~~~~^~
../main.cpp:2088:50: note: in expansion of macro ￢ﾀﾘVL_RANDOM_I_WIDTH￢ﾀﾙ
 2088 |                     top->iBus_rsp_payload_inst = VL_RANDOM_I_WIDTH(32);
      |                                                  ^~~~~~~~~~~~~~~~~
In file included from /usr/local/share/verilator/include/verilated_heavy.h:27,
                 from ./VVexRiscv.h:11,
                 from ../main.cpp:1:
/usr/local/share/verilator/include/verilated.h:935:14: note: declared here
[... cut ...] o-unused-result -DTHREAD_COUNT=1 -O3 -DCOREMARK -DWITH_RISCV_REF -DTIMER_INTERRUPT -DEXTERNAL_INTERRUPT -DDHRYSTONE -DSTALL=1 -DSEED=1526284434 -DTRACE -DCSR -DCSR_SKIP_TEST -DLRSC -DAMO -DISA_TEST -DMUL -DDIV -DTRACE_START=100000000000ll -DFREERTOS -DFREERTOS_COUNT=0 -DZEPHYR -DZEPHYR_COUNT=4  -std=gnu++14 -Os -c -o main.o ../main.cppmake[1]: Leaving directory '/home/vboxuser/test/VexRiscv/simWorkspace/test_id_99_/obj_dir'" did not contain "REGRESSION SUCCESS" (TestIndividualFeatures.scala:782)

Dolu1990, 2023-08-09:
@siatzjs What version of verilator ? it seems you may have a too old one.

siatzjs, 2023-08-10:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    Dear Dolu 1990:         Thank you very much for your reply, the version of Verilator is 4.210 2021-07-07
    
        
            
        
    
    
      
    
        
                    
            
                
                    
                            
                                
                            
                            
                                张佳帅
                            
                    
                        
                            
                                    ***@***.***
                            
                        
                
            
        
        
    
 
    

  
    ---- Replied Message ----
  
  
    
      
        
         From 
        
        
            ***@***.***>
            
        
      
      
        
         Date 
        
        
        08/10/2023 02:48
        
      
      
        
         To 
        
        
         
          
            ***@***.***>
            
          
        
      
      
        
         Cc 
        
        
          
            ***@***.***>
            ,
          
          
            ***@***.***>
            
          
        
      
      
        
         Subject 
        
        
              Re: [SpinalHDL/VexRiscv] verilator 5 - simulation failed for Linux.scala (Issue #287)
        
      
    
  
  
@siatzjs What version of verilator ? it seems you may have a too old one.

siatzjs, 2023-08-10:
> @siatzjs What version of verilator ? it seems you may have a too old one.
you are right,I tried using git pull to update the verilator,it can run now
thank you very much