SpinalHDL/VexRiscv #290: Error when trying to run linux in simulation

hparpinel, 2022-12-21:
Hi, I am trying to run Linux in simulation. I followed the instructions in the linux.gen file but I get the following output.
INUX_SOC=yes EMULATOR=../../../main/c/emulator/build/emulator.bin VMLINUX=$BUILDROOT/output/images/Image DTB=$BUILDROOT/board/spinal/vexriscv_sim/rv32.dtb RAMDISK=$BUILDROOT/output/images/rootfs.cpio WITH_USER_IO=yes TRACE=no FLOW_INFO=no
rm -rf obj_dir
rm -f VexRiscv.v*.bin
rm -f VexRiscv.v*.bin
cp ../../../../VexRiscv.v*.bin . | true
verilator -cc  ../../../../VexRiscv.v  -O3 -CFLAGS -std=c++11 -LDFLAGS -pthread  -CFLAGS -DIBUS_CACHED -CFLAGS -DDBUS_CACHED -CFLAGS -DREDO=0 -CFLAGS -pthread -CFLAGS -DTHREAD_COUNT=4 -CFLAGS -O3  -O3 -CFLAGS -DLINUX_SOC -CFLAGS -DVMLINUX='\"/home/bluespec/buildroot/output/images/Image\"' -CFLAGS -DDTB='\"/home/bluespec/buildroot/board/spinal/vexriscv_sim/rv32.dtb\"' -CFLAGS -DRAMDISK='\"/home/bluespec/buildroot/output/images/rootfs.cpio\"' -CFLAGS -DEMULATOR='\"../../../main/c/emulator/build/emulator.bin\"' -CFLAGS -DTIMER_INTERRUPT -CFLAGS -DEXTERNAL_INTERRUPT -CFLAGS -DWITH_USER_IO=yes -CFLAGS -DSUPERVISOR -CFLAGS -DSTALL=1 -CFLAGS -DCSR -CFLAGS -DLRSC -CFLAGS -DAMO -CFLAGS -DISA_TEST -CFLAGS -DMMU -CFLAGS -DMUL -CFLAGS -DDIV -CFLAGS -DTRACE_START=0 --gdbbt  -Wno-UNOPTFLAT -Wno-WIDTH --x-assign unique --exe main.cpp
No stack.
make  -j4 -C obj_dir/ -f VVexRiscv.mk VVexRiscv
make[1]: Entering directory '/home/bluespec/VexRiscv/src/test/cpp/regression/obj_dir'
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -DIBUS_CACHED -DDBUS_CACHED -DREDO=0 -pthread -DTHREAD_COUNT=4 -O3 -DLINUX_SOC -DVMLINUX=\"/home/bluespec/buildroot/output/images/Image\" -DDTB=\"/home/bluespec/buildroot/board/spinal/vexriscv_sim/rv32.dtb\" -DRAMDISK=\"/home/bluespec/buildroot/output/images/rootfs.cpio\" -DEMULATOR=\"../../../main/c/emulator/build/emulator.bin\" -DTIMER_INTERRUPT -DEXTERNAL_INTERRUPT -DWITH_USER_IO=yes -DSUPERVISOR -DSTALL=1 -DCSR -DLRSC -DAMO -DISA_TEST -DMMU -DMUL -DDIV -DTRACE_START=0  -std=gnu++14 -Os -c -o main.o ../main.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -DIBUS_CACHED -DDBUS_CACHED -DREDO=0 -pthread -DTHREAD_COUNT=4 -O3 -DLINUX_SOC -DVMLINUX=\"/home/bluespec/buildroot/output/images/Image\" -DDTB=\"/home/bluespec/buildroot/board/spinal/vexriscv_sim/rv32.dtb\" -DRAMDISK=\"/home/bluespec/buildroot/output/images/rootfs.cpio\" -DEMULATOR=\"../../../main/c/emulator/build/emulator.bin\" -DTIMER_INTERRUPT -DEXTERNAL_INTERRUPT -DWITH_USER_IO=yes -DSUPERVISOR -DSTALL=1 -DCSR -DLRSC -DAMO -DISA_TEST -DMMU -DMUL -DDIV -DTRACE_START=0  -std=gnu++14 -Os -c -o verilated.o /usr/local/share/verilator/include/verilated.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -DIBUS_CACHED -DDBUS_CACHED -DREDO=0 -pthread -DTHREAD_COUNT=4 -O3 -DLINUX_SOC -DVMLINUX=\"/home/bluespec/buildroot/output/images/Image\" -DDTB=\"/home/bluespec/buildroot/board/spinal/vexriscv_sim/rv32.dtb\" -DRAMDISK=\"/home/bluespec/buildroot/output/images/rootfs.cpio\" -DEMULATOR=\"../../../main/c/emulator/build/emulator.bin\" -DTIMER_INTERRUPT -DEXTERNAL_INTERRUPT -DWITH_USER_IO=yes -DSUPERVISOR -DSTALL=1 -DCSR -DLRSC -DAMO -DISA_TEST -DMMU -DMUL -DDIV -DTRACE_START=0  -std=gnu++14 -Os -c -o verilated_dpi.o /usr/local/share/verilator/include/verilated_dpi.cpp
/usr/bin/perl /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VVexRiscv.cpp VVexRiscv___024root__DepSet_hcd8610ab__0.cpp VVexRiscv___024root__DepSet_h0c12cad5__0.cpp VVexRiscv_VexRiscv__DepSet_hd894e8a5__0.cpp VVexRiscv_VexRiscv__DepSet_hd4e48a5e__0.cpp VVexRiscv__Dpi.cpp VVexRiscv__ConstPool_0.cpp VVexRiscv___024root__Slow.cpp VVexRiscv___024root__DepSet_hcd8610ab__0__Slow.cpp VVexRiscv___024root__DepSet_h0c12cad5__0__Slow.cpp VVexRiscv_VexRiscv__Slow.cpp VVexRiscv_VexRiscv__DepSet_hd894e8a5__0__Slow.cpp VVexRiscv_VexRiscv__DepSet_hd4e48a5e__0__Slow.cpp VVexRiscv__Syms.cpp > VVexRiscv__ALL.cpp
echo "" > VVexRiscv__ALL.verilator_deplist.tmp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -DIBUS_CACHED -DDBUS_CACHED -DREDO=0 -pthread -DTHREAD_COUNT=4 -O3 -DLINUX_SOC -DVMLINUX=\"/home/bluespec/buildroot/output/images/Image\" -DDTB=\"/home/bluespec/buildroot/board/spinal/vexriscv_sim/rv32.dtb\" -DRAMDISK=\"/home/bluespec/buildroot/output/images/rootfs.cpio\" -DEMULATOR=\"../../../main/c/emulator/build/emulator.bin\" -DTIMER_INTERRUPT -DEXTERNAL_INTERRUPT -DWITH_USER_IO=yes -DSUPERVISOR -DSTALL=1 -DCSR -DLRSC -DAMO -DISA_TEST -DMMU -DMUL -DDIV -DTRACE_START=0  -std=gnu++14 -Os -c -o VVexRiscv__ALL.o VVexRiscv__ALL.cpp
../main.cpp: In member function ‘virtual void Workspace::dBusAccess(uint32_t, bool, uint32_t, uint32_t, uint32_t*, bool*)’:
../main.cpp:1336:27: error: too many arguments to function ‘IData VL_RANDOM_I()’
 1336 |     *data = VL_RANDOM_I(32);
      |                           ^
In file included from /usr/local/share/verilator/include/verilated.h:881,
                 from ./VVexRiscv.h:11,
                 from ../main.cpp:1:
/usr/local/share/verilator/include/verilated_funcs.h:77:14: note: declared here
   77 | inline IData VL_RANDOM_I() VL_MT_SAFE { return vl_rand64(); }
      |              ^~~~~~~~~~~
../main.cpp: At global scope:
../main.cpp:1772:17: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
 1772 |  char *target = "PROJECT EXECUTION SUCCESSFUL", *hit = target;
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../main.cpp: In member function ‘virtual void ZephyrRegression::dutPutChar(char)’:
../main.cpp:1781:20: warning: NULL used in arithmetic [-Wpointer-arith]
 1781 |         if(*hit == NULL) {
      |                    ^~~~
../main.cpp: In member function ‘virtual void IBusCached::postCycle()’:
../main.cpp:1958:56: error: too many arguments to function ‘IData VL_RANDOM_I()’
 1958 |   if(pendingCount != 0 && (!ws->iStall || VL_RANDOM_I(7) < 100)){
      |                                                        ^
In file included from /usr/local/share/verilator/include/verilated.h:881,
                 from ./VVexRiscv.h:11,
                 from ../main.cpp:1:
/usr/local/share/verilator/include/verilated_funcs.h:77:14: note: declared here
   77 | inline IData VL_RANDOM_I() VL_MT_SAFE { return vl_rand64(); }
      |              ^~~~~~~~~~~
../main.cpp:1971:53: error: too many arguments to function ‘IData VL_RANDOM_I()’
 1971 |   if(ws->iStall) top->iBus_cmd_ready = VL_RANDOM_I(7) < 100 && pendingCount == 0;
      |                                                     ^
In file included from /usr/local/share/verilator/include/verilated.h:881,
                 from ./VVexRiscv.h:11,
                 from ../main.cpp:1:
/usr/local/share/verilator/include/verilated_funcs.h:77:14: note: declared here
   77 | inline IData VL_RANDOM_I() VL_MT_SAFE { return vl_rand64(); }
      |              ^~~~~~~~~~~
../main.cpp: In member function ‘virtual void DBusCached::postCycle()’:
../main.cpp:2272:63: error: too many arguments to function ‘IData VL_RANDOM_I()’
 2272 |   if(pendingCount != 0 && !wr && (!ws->dStall || VL_RANDOM_I(7) < 100)){
      |                                                               ^
In file included from /usr/local/share/verilator/include/verilated.h:881,
                 from ./VVexRiscv.h:11,
                 from ../main.cpp:1:
/usr/local/share/verilator/include/verilated_funcs.h:77:14: note: declared here
[... cut ...]
../main.cpp:2820:11: warning: ignoring return value of ‘size_t fread(void*, size_t, size_t, FILE*)’, declared with attribute warn_unused_result [-Wunused-result]
 2820 |      fread(log, 1, logSize, logFile);
      |      ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
make[1]: *** [VVexRiscv.mk:87: main.o] Error 1
make[1]: *** Waiting for unfinished jobs....
rm VVexRiscv__ALL.verilator_deplist.tmp
make[1]: Leaving directory '/home/bluespec/VexRiscv/src/test/cpp/regression/obj_dir'
make: *** [makefile:272: compile] Error 2

Dolu1990, 2022-12-22:
Hi, 
What version of VexRiscv do you have ? i think that issue has been fixed.
(error: too many arguments to function ‘IData VL_RANDOM_I()’)

hparpinel, 2022-12-22:
I updated my version but now I am getting this error: ./obj_dir/VVexRiscv
BOOT
Liveness step failure
FAIL linux at PC=80000000 REF PC=0 REF I=0 time=20016
make: *** [makefile:345: run] Error 255

Dolu1990, 2022-12-22:
-b linux   is very very outdated
 
 git clone https://github.com/SpinalHDL/VexRiscv.git --recursive should be good

hparpinel, 2022-12-22:
Thank you for your reply. I have cloned the repository using the command  $ git clone https://github.com/SpinalHDL/VexRiscv.git --recursive
However, when I try to to compile the machine mode emulator I get the following error:
$ cd src/main/c/emulator
$ make clean all
rm -f build/emulator.elf
rm -f build/emulator.hex
rm -f build/emulator.map
rm -f build/emulator.v
rm -f build/emulator.bin
rm -f build/emulator.asm
find build -type f -name '*.o' -print0 | xargs -0 -r rm
mkdir -p build/src/
riscv64-unknown-elf-gcc -c -march=rv32i -mabi=ilp32 -DUSE_GP -I../include -O3   -o build/src/main.o src/main.c
src/main.c: In function 'trap':
src/main.c:182:22: error: 'instruction' undeclared (first use in this function)
    uint32_t opcode = instruction & 0x7F;
                      ^~~~~~~~~~~
src/main.c:182:22: note: each undeclared identifier is reported only once for each function it appears in
make: *** [../common/standalone.mk:52: build/src/main.o] Error 1

This are the steps I am following to run Linux in simulation is there something I am missing?

== INSTALL SPINALHDL AND VEXRISCV
$ git clone https://github.com/SpinalHDL/SpinalHDL.git -b dev
$ git clone https://github.com/SpinalHDL/VexRiscv.git --recursive 

== TOOLCHAIN AND EMULATOR
$ version=riscv64-unknown-elf-gcc-8.3.0-2019.08.0-x86_64-linux-ubuntu14
wget -O riscv64-unknown-elf-gcc.tar.gz riscv https://static.dev.sifive.com/dev-tools/$version.tar.gz
$ tar -xzvf riscv64-unknown-elf-gcc.tar.gz
$ sudo mv $version /opt/riscv
$ echo 'export PATH=/opt/riscv/bin:$PATH' >> ~/.bashrc

== BUILDROOT
$ git clone https://github.com/SpinalHDL/buildroot.git -b vexriscv
$ cd buildroot
$ make spinal_vexriscv_sim_defconfig
$ make -j$(nproc)
$ output/host/bin/riscv32-linux-objcopy  -O binary output/images/vmlinux output/images/Image

== COMPILE MACHINE MODE EMULATOR
$ src/main/c/emulator
$ make clean all

== RUN LINUX IN SIMULATION
$  "runMain vexriscv.demo.LinuxGen"
$ cd src/test/cpp/regression
$ export BUILDROOT=/path/to/buildroot
$ make clean run IBUS=CACHED DBUS=CACHED  DEBUG_PLUGIN=STD SUPERVISOR=yes CSR=yes DEBUG_PLUGIN=no  COMPRESSED=no LRSC=yes AMO=yes REDO=0 DHRYSTONE=no LINUX_SOC=yes EMULATOR=../../../main/c/emulator/build/emulator.bin VMLINUX=$BUILDROOT/output/images/Image DTB=$BUILDROOT/board/spinal/vexriscv_sim/rv32.dtb RAMDISK=$BUILDROOT/output/images/rootfs.cpio WITH_USER_IO=yes TRACE=no FLOW_INFO=no

Dolu1990, 2022-12-23:
Hmmm, try just : 
make clean
make

Worked for me.

hparpinel, 2022-12-23:
That worked but I am still getting the error below when I run 
$ make clean run IBUS=CACHED DBUS=CACHED  DEBUG_PLUGIN=STD SUPERVISOR=yes CSR=yes DEBUG_PLUGIN=no  COMPRESSED=no LRSC=yes AMO=yes REDO=0 DHRYSTONE=no LINUX_SOC=yes EMULATOR=../../../main/c/emulator/build/emulator.bin VMLINUX=$BUILDROOT/output/images/Image DTB=$BUILDROOT/board/spinal/vexriscv_sim/rv32.dtb RAMDISK=$BUILDROOT/output/images/rootfs.cpio WITH_USER_IO=yes TRACE=no FLOW_INFO=no

ERROR:
./obj_dir/VVexRiscv
BOOT
Liveness step failure
FAIL linux at PC=80000000 REF PC=0 REF I=0 time=20016
make: *** [makefile:345: run] Error 255

Dolu1990, 2023-01-02:
Ahhh, so, one issue you got is that DEBUG_PLUGIN=no  should be DEBUG_PLUGIN=STD in your command line, because the debug plugin was added since.

Here is how i just got it to run : 
```shell
export ARCH_LINUX=rv32ima
export VMLINUX=../../resources/VexRiscvRegressionData/sim/linux/${ARCH_LINUX}/Image
export DTB=../../resources/VexRiscvRegressionData/sim/linux/${ARCH_LINUX}/rv32.dtb
export RAMDISK=../../resources/VexRiscvRegressionData/sim/linux/${ARCH_LINUX}/rootfs.cpio
export EMULATOR=../../resources/VexRiscvRegressionData/sim/linux/emulator/emulator.bin

make clean run IBUS=CACHED DBUS=CACHED  DEBUG_PLUGIN=STD SUPERVISOR=yes CSR=yes DEBUG_PLUGIN=STD  COMPRESSED=no LRSC=yes AMO=yes REDO=0 DHRYSTONE=no LINUX_SOC=yes WITH_USER_IO=yes TRACE=no FLOW_INFO=no
```