SpinalHDL/VexRiscv #293: Regression Tests not passed?

Light9162, 2023-01-05:
Duration=4mn clocks=375M clockPerSecond=1269K
[info] - report
[info] Run completed in 4 minutes, 56 seconds.
[info] Total number of tests run: 102
[info] Suites: completed 1, aborted 0
[info] Tests: succeeded 15, failed 87, canceled 0, ignored 0, pending 0
[info] *** 87 TESTS FAILED ***
[error] Failed tests:
[error] 	vexriscv.TestIndividualFeatures
[error] (Test / testOnly) sbt.TestsFailedException: Tests unsuccessful
[error] Total time: 304 s (05:04), completed Jan 5, 2023 2:08:17 PM

I compiled the full core and ran the full regression suite, are all the tests suppose to pass?

Dolu1990, 2023-01-06:
Hi
Did you cloned the repo recursively ? it has git submodules.

Light9162, 2023-01-06:
Caught me! That fixed it!

For others reference:
WSL (Windows 11 + Ubuntu 20.04):
Helpful Information:
Installation WSL: https://learn.microsoft.com/en-us/windows/wsl/install
Cloning Repos: https://www.theserverside.com/blog/Coffee-Talk-Java-News-Stories-and-Opinions/How-to-git-clone-on-Ubuntu-with-GitLab-and-GitHub

Install dependencies (Do not be connected to the VPN while doing this): 
# JAVA JDK 8
sudo add-apt-repository -y ppa:openjdk-r/ppa
sudo apt-get update
sudo apt-get install openjdk-8-jdk -y
# Run the following commands, if you have multiple alternatives from previous installations switch to      # JAVA 8 (we just installed it)
sudo update-alternatives --config java
sudo update-alternatives --config javac

# Install SBT - https://www.scala-sbt.org/
echo "deb https://repo.scala-sbt.org/scalasbt/debian all main" | sudo tee /etc/apt/sources.list.d/sbt.list
echo "deb https://repo.scala-sbt.org/scalasbt/debian /" | sudo tee /etc/apt/sources.list.d/sbt_old.list
curl -sL "https://keyserver.ubuntu.com/pks/lookup?op=get&search=0x2EE0EA64E40A89B84B2DF73499E82A75642AC823" | sudo apt-key add
sudo apt-get update
sudo apt-get install sbt

# Verilator (for sim only, really needs 3.9+, in general apt-get will give you 3.8)
sudo apt-get install git make autoconf g++ flex bison
git clone http://git.veripool.org/git/verilator   # Only first time
unsetenv VERILATOR_ROOT  # For csh; ignore error if on bash
unset VERILATOR_ROOT  # For bash
cd verilator
git pull        # Make sure we're up-to-date
git checkout v4.216
autoconf        # Create ./configure script
./configure
make # This will take some time. 
sudo make install

Clone the repo and generate the verilog:
git clone --recurse-submodules https://github.com/SpinalHDL/VexRiscv.git
cd VexRiscv
sbt "runMain vexriscv.demo.GenFull"

Run Regressions:
export VEXRISCV_REGRESSION_SEED=42
export VEXRISCV_REGRESSION_TEST_ID=
sbt "testOnly vexriscv.TestIndividualFeatures" #Run Regressions, should take 30 minutes or so and all 102 tests should pass

closed 2023-01-06 by Light9162

Dolu1990, 2023-01-09:
Thanks for the reference :)