SpinalHDL/VexRiscv #295: DE1-SOC Quartus Full design synthesis issue

Light9162, 2023-01-13:
Hello! @Dolu1990, I know you previously answered a similar question on issue #22. I running into the same error on the DE1-SOC which uses a Cyclone V chip. My confusion comes from the fact that you have benchmark numbers for the full chip on Cyclone V so I would think it would work out of the box no? Should I try to cut GPIO pins from the design also? What file do you suggested I edit to do that?

Thank you in advanced!

[image]

Dolu1990, 2023-01-13:
Hi, the pure VexRiscv synthesis number are with FPGA with many many pins and without the SoC infrastructure :
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/SynthesisBench.scala

If you want a SoC, you can check Murax / Briey, which have everything for a standalone SoC.

nerdylye, 2024-12-17:
hello, how do you run the synthesis from this 
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/SynthesisBench.scala ?

Dolu1990, 2024-12-19:
Hi,

sbt "runMain vexriscv.demo.VexRiscvSynthesisBench"  to run it, but you need first to define a few environnement variables. For instance, me i have : 
```shell
export QUARTUS_CYCLONE_IV_BIN="/media/data2/tools/intelFPGA/23.1std/quartus/bin"
export QUARTUS_CYCLONE_V_BIN="/media/data2/tools/intelFPGA/23.1std/quartus/bin"
export VIVADO_ARTIX_7_BIN="/media/data2/tools/xilinx/Vivado/2023.1/bin"
export EFINIX_BIN="/media/data2/tools/2023.2/bin"
```