SpinalHDL/VexRiscv #299: Issues using Eclipse to run and debug the software

hparpinel, 2023-02-13:
Hi,

I am trying to use Eclipse to run and debug the software following the steps in the repository but I get the error below. I installed the XPacks and added the path to openocd. I am not sure what I am missing.

Error in final launch sequence
Failed to execute MI command:
-target-select remote localhost:3333
Error message from debugger back end:
localhost:3333: Connection timed out.
Failed to execute MI command:
-target-select remote localhost:3333
Error message from debugger back end:
localhost:3333: Connection timed out.
localhost:3333: Connection timed out.

Dolu1990, 2023-02-15:
Hi,

What's is openocd logging in the shell when it is running ?

acillatems, 2023-02-24:
Hi!

I've got the same error. Here are the only errors in openOCD logging:

342,122 13-gdb-set charset ISO-8859-1
342,138 13^error,msg="Undefined item: \"ISO-8859-1\"."
342,138 (gdb) 
342,138 14source .gdbinit
342,154 &"source .gdbinit\n"
342,154 &".gdbinit: No such file or directory.\n"
342,154 14^error,msg=".gdbinit: No such file or directory."
342,154 (gdb)

Dolu1990, 2023-02-27:
Hi,

I don't mean GDB, but realy openocd itself, GDB is the debugger, while openocd is the link between GDB and the physical layer connecting the target. (GDB is the client, openocd is the server)

acillatems, 2023-02-27:
Hi,

Oh my bad. I am on Windows and when I try to run openocd with the right .cfg files I get this error: "unable to open ftdi device with vid 0403, pid 6014, description 'Digilent USB Device', serial '*' at bus location '*'". I have managed to generate MuraxWithRamInit and programmed that on my device but I can't seem to be able to debug and run the software on eclipse and therefore on to my device.

Dolu1990, 2023-02-28:
hmm in the config file, maybe remove the "Digilent USB Device"  setting.
can you show what file you are using ?

acillatems, 2023-03-01:
I assume the problem is because I am not using the openocd_riscv. The problem with that is that I don't really know how to build that on windows easily.