SpinalHDL/VexRiscv #305: Problem with the `fence` instruction

flaviens, 2023-02-26:
Hi there!

I reverse-engineered a bit the BMB protocol and set up some very simple SoC in simulation that seems to work fine with simple arithmetic and memory operations. To simplify things, I decided to have cache lines as wide as the memory busses for now.
However, if I run a `fence` instruction, then subsequent memory operations do not happen. For example if I try to do a store, then I see some address signals being set up, but the valid signal is never raised. I'm running everythin in machine mode for now.
I tried to add the `DummyFencePlugin` plugin (which is instantiated nowhere apparently) but it did not help.

The configuration that I am using is close to the examples, with a 64-bit bus to support double-precision FPU in the future.

```
object VexRiscvAhbLite3cellift{
  def main(args: Array[String]) {
    val report = SpinalConfig(mode = if(args.contains("--vhdl")) VHDL else Verilog).generate{

      //CPU configuration
      val cpuConfig = VexRiscvConfig(
        plugins = List(
          // new IBusSimplePlugin(
          //   resetVector = 0x80000000l,
          //   cmdForkOnSecondStage = false,
          //   cmdForkPersistence = true,
          //   prediction = DYNAMIC_TARGET,
          //   catchAccessFault = true,
          //   compressedGen = true
          // ),
          // new DBusSimplePlugin(
          //   catchAddressMisaligned = true,
          //   catchAccessFault = true
          // ),
          new IBusCachedPlugin(
            resetVector = 0x80000000l,
            compressedGen = false, // TODO
            prediction = STATIC, // TODO
            historyRamSizeLog2 = 4,
            injectorStage = false,
            config = InstructionCacheConfig(
              cacheSize = 4096*1,
              bytePerLine = 8,
              wayCount = 2,
              addressWidth = 32,
              cpuDataWidth = 32,
              memDataWidth = 64,
              catchIllegalAccess = true,
              catchAccessFault = true,
              asyncTagMemory = false,
              twoCycleRam = false,
              twoCycleCache = true
            ),
            memoryTranslatorPortConfig = MmuPortConfig(
              portTlbSize = 4
            )
          ),
          new DBusCachedPlugin(
            dBusCmdMasterPipe = true,
            dBusCmdSlavePipe = true,
            dBusRspSlavePipe = true,
            config = new DataCacheConfig(
              cacheSize         = 4096*1,
              bytePerLine       = 8,
              wayCount          = 2,
              addressWidth      = 32,
              cpuDataWidth      = 64,
              memDataWidth      = 64,
              catchAccessError  = true,
              catchIllegal      = true,
              catchUnaligned    = true,
              withExclusive = false, // = withsmp
              withInvalidate = false, // = withsmp
              withLrSc = false,
              withAmo = false,
              withWriteAggregation = true //dBusWidth > 32
  //          )
            ),
            memoryTranslatorPortConfig = MmuPortConfig(
              portTlbSize = 4
            )
          ),
          new MmuPlugin(
            virtualRange = _(31 downto 28) === 0xC,
            ioRange      = _(31 downto 28) === 0xF
          ),
          new FpuPlugin(
            externalFpu = false,
            simHalt = false,
            p = FpuParameter(withDouble = true)
          ),
          new DecoderSimplePlugin(
            catchIllegalInstruction = true
          ),
          new RegFilePlugin(
            regFileReadyKind = plugin.SYNC,
            zeroBoot = false
          ),
          new IntAluPlugin,
          new SrcPlugin(
            separatedAddSub = false,
            executeInsertion = true
          ),
          new FullBarrelShifterPlugin,
          new MulPlugin,
          new DivPlugin,
          new HazardSimplePlugin(
            bypassExecute           = true,
            bypassMemory            = true,
            bypassWriteBack         = true,
            bypassWriteBackBuffer   = true,
            pessimisticUseSrc       = false,
            pessimisticWriteRegFile = false,
            pessimisticAddressMatch = false
          ),
          new BranchPlugin(
            earlyBranch = false,
            catchAddressMisaligned = true
          ),
          new CsrPlugin(
            config = CsrPluginConfig(
              catchIllegalAccess = false,
              mvendorid      = null,
              marchid        = null,
              mimpid         = null,
              mhartid        = null,
              misaExtensionsInit = 66,
              misaAccess     = CsrAccess.NONE,
              mtvecAccess    = CsrAccess.NONE,
              mtvecInit      = 0x00000020l,
              mepcAccess     = CsrAccess.READ_WRITE,
              mscratchGen    = false,
              mcauseAccess   = CsrAccess.READ_ONLY,
              mbadaddrAccess = CsrAccess.READ_ONLY,
              mcycleAccess   = CsrAccess.NONE,
              minstretAccess = CsrAccess.READ_ONLY,
              ecallGen       = true,
              ebreakGen      = true,
              wfiGenAsWait   = false,
              ucycleAccess   = CsrAccess.NONE,
              uinstretAccess = CsrAccess.READ_ONLY
            )
          ),
          new YamlPlugin("cpu0.yaml")
        )
      )
      //CPU instanciation
      val cpu = new VexRiscv(cpuConfig)
      cpu
    }
  }
}
```

A software as simple as
```
fence
sw x0, (x0)
```
does not store. But without the fence it works.

Is there a problem with my configuration or is fence not supported in that case?
I remain at your disposal, may you need more info.

Thanks!
Flavien

Dolu1990, 2023-02-26:
Hi, 

Hmm if you take the CPU directly as generated by your code, it would not provide a BMB bus, but something native to the CPU. Then some bridge can be applied to go to less VexRiscv specific busses.

I would say keep  withWriteAggregation : to false.

I think you don't need the DummyFencePlugin, as fence is implemented via https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/DBusCachedPlugin.scala#L236

I would say, if you can share a wave file of the blocked system, that could help ^^

flaviens, 2023-02-26:
Hi @Dolu1990 thank you for your super quick answer!
I'll try the parameter you said. Let's try to save you some waveform reading but in the worst case we can do that :)

At the beginning I used a derivative of the AHBLite config and would be happy if I could continue with that. However, I wanted to add MMU and FPUd and it was not clear whether I could easily combine that with the the AHBLite (probably technically it should not be problematic I hope). If yes, then that would resolve the whole problem. Do you have an example configuration with FPUd and MMU, bridged to AHB Lite?
Thank you very much!
Flavien

flaviens, 2023-02-27:
Hi! [Here are three VCDs](https://github.com/SpinalHDL/VexRiscv/files/10838106/vcds.zip):
nofencenonop.vcd:
```
sw x0, (x0)
infinite_loop:
  j infinite_loop
```
nofencevcd:
```
nop
sw x0, (x0)
infinite_loop:
  j infinite_loop
```
fencevcd:
```
fence
sw x0, (x0)
infinite_loop:
  j infinite_loop
```

Only `nofencenonop.vcd` actually writes to the memory, which suggests that the fence is not the problem itself. Probably I did not understand the internal bus correctly.

Dolu1990, 2023-02-27:
Hi, 

Thanks for the wave,
So the issue seems to come outside VexRiscv, the ibus receive 3 beat in response from one read : 
[image]

That mess up things in VexRiscv then ^^

flaviens, 2023-02-27:
Thank you for your super fast response! Oh how can I be so stupid :facepalm: I'll fix this and let you know!

flaviens, 2023-02-27:
Ah yes now I remember! This is because I saw `iBus_cmd`payload_size` which I thought would correspond to BMB's `length`, but probably not, then. What does this `size` mean?
Looking in the Vexriscv sources I saw it's something around `log2up(log2up(linewidth)+1)`. How  should I interpret this value, or should I ignore it at all?
Thanks!

Dolu1990, 2023-02-27:
So : 
io.mem.cmd.size := log2Up(p.bytePerLine)
3  => 1 << 3  = 8 bytes 
6 => 1 << 6 = 64 bytes

flaviens, 2023-02-27:
This I follow :) but what is it used for?

Dolu1990, 2023-02-27:
right, that's not linear, and in bytes
Sorry, which "last" signal ?

flaviens, 2023-02-27:
These are the signals for the dbus and ibus

```
  output              dBus_cmd_valid,
  input               dBus_cmd_ready,
  output              dBus_cmd_payload_wr,
  output              dBus_cmd_payload_uncached,
  output     [31:0]   dBus_cmd_payload_address,
  output     [63:0]   dBus_cmd_payload_data,
  output     [7:0]    dBus_cmd_payload_mask,
  output     [1:0]    dBus_cmd_payload_size,
  output              dBus_cmd_payload_last,
  input               dBus_rsp_valid,
  input               dBus_rsp_payload_last,
  input      [63:0]   dBus_rsp_payload_data,
  input               dBus_rsp_payload_error,
  output              iBus_cmd_valid,
  input               iBus_cmd_ready,
  output reg [31:0]   iBus_cmd_payload_address,
  output     [1:0]    iBus_cmd_payload_size,
  input               iBus_rsp_valid,
  input      [63:0]   iBus_rsp_payload_data,
  input               iBus_rsp_payload_error,
```
I was referring to the `dBus_cmd_payload_last` / `dBus_rsp_payload_last` pair.

Dolu1990, 2023-02-27:
Ahhh, there is a last signal on dbus because the number of beat varies, while ibus always do 1 cache line access.

flaviens, 2023-02-27:
Alright! That makes sense, then it implies that I can ignore `iBus_cmd_payload_size`?

Dolu1990, 2023-02-27:
yes

flaviens, 2023-02-27:
Hi @Dolu1990,
I think all issues related to the bus are now gone, thank you so much for your efficient support! Extremely appreciated! :pray: :slightly_smiling_face:

closed 2023-02-27 by flaviens