SpinalHDL/VexRiscv #308: How Can I support the Daisy-chain debugging

xie-1399, 2023-02-27:
Hello，if I want to support the Daisy-chain debugging with the vexriscv ,  what should I do ? can you show me some examples , Thank you very much! 

what's more , When i excute src/openocd -c "set VEXRISCV_YAML PATH_TO_THE_GENERATED_CPU0_YAML_FILE" -f tcl/target/vexriscv_sim.cfg , there are some mistakes
[image]

Dolu1990, 2023-02-27:
Hi,

So daisy chain multiple JTAG ? Is your screen shot from a simulated target ? or a real target (can see vexriscv_sim.cfg)

xie-1399, 2023-02-27:
Hi,

I want to implement daisy chain with multiple vexriscv cpu and we can debug these cpus by JATG ， Maybe like this
 
[image]

Dolu1990, 2023-02-27:
Ahh you want to debug multiple VexRiscv CPU that are in the same device ?

xie-1399, 2023-02-27:
Hi,

In the different devices and can Connect them by Jtag ，Maybe this picture is more appropriate
[image]

xie-1399, 2023-02-27:
In fact each device have one vexriscv cpu，But I don't know how to achieve this chain

Dolu1990, 2023-02-27:
So, as it is in your last picture is right :)

Dolu1990, 2023-02-27:
Thing is you will need to adapte the openocd script to specify each tap on the chain, starting from the device 0

xie-1399, 2023-02-27:
yeah ! AndI think Vexriscv can support this chain(*^_^*)

xie-1399, 2023-02-27:
I try to use the openocd script like this , But It can't work fine

interface dummy
set  _ENDIAN little
set _TAP_TYPE 1234

if { [info exists CPUTAPID] } {
   set _CPUTAPID $CPUTAPID
} else {
  # set useful default
   set _CPUTAPID 0x10001fff
}
adapter_khz 4000
adapter_nsrst_delay 260
jtag_ntrst_delay 250
transport select jtag

set _CHIPNAME fpga_spinal
set _CHIP1 riscv_1
set _CHIP0 riscv_0

jtag newtap $_CHIP0 bridge -expected-id 0x80022001 -irlen 4 -ircapture 0x1 -irmask 0x1F
jtag newtap $_CHIP1 bridge -expected-id $_CPUTAPID -irlen 4 -ircapture 0x1 -irmask 0x1F
#Add Jtag Tap

target create $_CHIP0 vexriscv -endian $_ENDIAN -chain-position $_CHIP0.bridge -coreid 0 -dbgbase 0xF00F0000
target create $_CHIP1 vexriscv -endian $_ENDIAN -chain-position $_CHIP1.bridge -coreid 1 -dbgbase 0xF00F0000

vexriscv readWaitCycles 12
vexriscv cpuConfigFile $VEXRISCV_YAML
vexriscv networkProtocol iverilog

Dolu1990, 2023-02-27:
I don't understand vexriscv networkProtocol iverilog

You are running a simulation ? or real hardware ?

xie-1399, 2023-02-27:
running a simulation，like the vexriscv_sim.cfg

Dolu1990, 2023-02-28:
could you share a wave file of the simulation when openocd try to connect ?

closed 2023-03-02 by xie-1399