SpinalHDL/VexRiscv #309: Bug report: wrong sign injection when encountering NaNs

flaviens, 2023-02-28:
Hi there!

I've detected a bug in VexRiscv.

## Brief bug description

Some sign injection instructions, such as `fsgnj.d` or `fsgnjx.d`, behave incorrectly when encountering NaNs.

## Example instance

### Snippet

Here is an example RISC-V (rv32imfd) snippet:
```
  .section ".text.init","ax",@progbits
  .globl _start
  .align 2
_start:

  # Enable the FPU
  li t0, 0x2000
  csrs mstatus, t0
  csrw	fcsr,x0

  la t0, .fdata0
  la t1, .fdata1

  fld f14, 0(t0)
  fld f3, 0(t1)
  fsgnj.d	f4,f14,f3

  li t2, 0x80000018
  fsd f4, 0(t2)

  sw x0, (x0)

infinite_loop:
  jal	x0,infinite_loop

.section ".fdata0","ax",@progbits
  .8byte 0xffffffff7fc00000
.section ".fdata1","ax",@progbits
  .8byte 0x7ff8000000000000
```

### Waves

Here is [the VCD trace](https://github.com/SpinalHDL/VexRiscv/files/10850173/fsgnj.zip) of the execution of this program.

### Expected and actual results

We expect `f4 = 0x7fffffff7fc00000` because the sign bit of `f3` is 0. I verified this with Spike. Other designs such as Rocket and BOOM provide this result as well.
However,  VexRiscv returns `f4 = 0xffffffff7fc00000`.

The specification says:
> Floating-point to floating-point sign-injection instructions, FSGNJ.S, FSGNJN.S, and FSGNJX.S, produce a result that takes all bits except the sign bit from rs1. For FSGNJ, the result’s sign bit is rs2’s sign bit [...] Sign-injection instructions do not set floating-point exception flags, nor do they canonicalize NaN

## Fix

I identified the area probably responsible for the sign injection, which is
https://github.com/SpinalHDL/VexRiscv/blob/b29eb542f278edb8ee2c91863a5b26e76b5932c9/src/main/scala/vexriscv/ip/fpu/FpuCore.scala#L797-L857
However, I am very unfamiliar with the design itself and with SpinalHDL.
If you provide some guidance, I would be happy to help.

Thanks!
Flavien

Dolu1990, 2023-02-28:
Hi ^^

It is a known behaviour "documented" as "Very specific, but SGNJ instruction will not mutate the value from/to F32/F64 (no NaN-boxing mutation)" in the main readme.md :)

I will have to take a look, i'm not sure if this is a bug which could create practical issues, i would say it isn't ?

Dolu1990, 2023-02-28:
> What would it cost to fix it?

Some headaches i think ^^
It was related to how the floatpoint was re-encoded internaly i think, it was a long time ago XD

On NaxRiscv (https://spinalhdl.github.io/NaxRiscv-Rtd/main/NaxRiscv/introduction/index.html) i got it right, mostly because there was no re-encoding

flaviens, 2023-02-28:
I'm willing to help! :slightly_smiling_face: if you have some indications I can try.