SpinalHDL/VexRiscv #310: Bug report: non-deterministic conversion from float to int

flaviens, 2023-02-28:
Hi there!

I've detected a bug (or feature that I could not find documented in [the Readme](https://github.com/SpinalHDL/VexRiscv/blob/b29eb542f278edb8ee2c91863a5b26e76b5932c9/README.md)?) in VexRiscv.

## Brief bug description

Some float to int instructions, such as `fcvt.wu.s`, provide non-deterministic outputs.
It seems that the FPU keeps some state beyond the precision stipulated in the standards.

## Example instance

### Snippet

Here is an example RISC-V (rv32imfd) snippet:
```
  .section ".text.init","ax",@progbits
  .globl _start
  .align 2
_start:
  # Enable the FPU
  li t0, 0x2000
  csrs mstatus, t0
  csrw	fcsr,x0

  li t5, 0xfffffd53
  li t6, 0xfffffd00

  fcvt.s.wu	ft3,t5
  fcvt.s.wu	ft4,t6

  # At this point, ft3 = ft4 = 0xffffffff4f7ffffd = 4.29497e+09 interpreted in simple precision.

  fcvt.wu.s	t3,ft3
  fcvt.wu.s	t4,ft4

  li t2, 0x80000010
  sw t3, 0(t2)
  sw t4, 0(t2)

  sw x0, (x0)

infinite_loop:
  jal	x0,infinite_loop
```

### Waves

Here is [the VCD trace](https://github.com/SpinalHDL/VexRiscv/files/10851047/fcvt.zip) of an execution of this program.

### Expected and actual results

We expect `t3 = t4 = 0xfffffd00`. I verified this with Spike. BOOM provides this result as well.
However,  VexRiscv returns `t4=0xfffffd00` but `t3 = 0xfffffd53`, while this originates from the same architectural floating-point representation as from IEEE 754-2019.

## Fix

Even though I'm very unfamiliar with Vexriscv and SpinalHDL, I'll be very happy to help fix this issue!

Thanks!
Flavien

Dolu1990, 2023-03-01:
https://github.com/SpinalHDL/VexRiscv/commit/5f67075e3094f1f31b7fe9f480e08fc5c32b28c3 Seems to fix it.

Dolu1990, 2023-03-01:
(note, for all the other issue that i closed with the https://github.com/SpinalHDL/VexRiscv/commit/5f67075e3094f1f31b7fe9f480e08fc5c32b28c3 Seems to fix it. comment, i tested in the wave manualy with your code ^^, it isn't just closed assuming)

closed 2023-03-01 by Dolu1990