SpinalHDL/VexRiscv #311: How to interface with a VHDL module with Axi4

corentin-fleury, 2023-02-28:
Hi, I have a module written in VHDL which has an Axi4 interface and I would like to interface it with the Axi4 class of the SpinalHDL so I can use it inside the Vex. My question is: what are the steps I need to follow to do this ? As of now I created a SpinalHDL wrapper around the VHDL module. Then I created a class which instantiates this module and now I am trying to connect the VHDL ports to the corresponding class ports:
```
class CustomModuleAxi() {

    val module = CustomModule() //Default param
    object axi_config = Axi4Config(
        dataWidth = 32,
        useId = False,
        useQos = False,
        useRegion = False,
        useLock = False,
        useCache = False,
        useBurst = False,
        useSize = False,
        useLen = False,
        useProt = False,
        useUser = False,
        useStrb = False,
        useResp = False
    )

    val io = Bundle(
        axi_slave = slave(Axi4Shared(axi_config))
    )

    io.axi_slave.writeRsp.ready := module.io.S_Tready_o
    io.axi_slave.writeRsp.valid := module.io.M_Tvalid_o
    io.axi_slave.writeRsp.last := module.io.M_Tlast_o
}
```

How can I say for instance that the writeData ports will be connected to the register-file ?

Dolu1990, 2023-03-01:
Hi,

Seems like you need to use the BlackBox feature : 
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Structuring/blackbox.html#instantiate-vhdl-and-verilog-ip

also, in the blackbox you can "normalize" the name of the axi signal via the Axi4SpecRenamer tool

val myAxiBus = Axi4SpecRenamer(slave(Axi4Shared(axi_config)))

corentin-fleury, 2023-03-01:
I looked at Axi4SpecRenamer but I don't fully understand how it maps signals. To me, it seems to work by making assumptions about the original signal name. But what if the signal aren't named exactly like it assumed them to be ? Isn't there a way to connect signals like you would do it in VHDL:
    port1=>sig1

Dolu1990, 2023-03-02:
> port1=>sig1

Yes, you can define and then connect every in out   one by one, you can also define a SpinalHDL AXI interface, and force the name of the signals one by one  (myAxi.ar.address.setName("rawrr")