SpinalHDL/VexRiscv #317: Bug report: strict single-precision floating point comparisons (flt.s) may return 1 even though the operands are equal

flaviens, 2023-02-28:
Hi there!

I've detected a bug in VexRiscv.

## Brief bug description

The instruction `flt.s` must return `0` when the two operands are equal. Under certain conditions, it returns `1` even though the operands are equal.

## Example instance

### Snippet

Here is an example RISC-V (rv32imfd) snippet:
```
  .section ".text.init","ax",@progbits
  .globl _start
  .align 2
_start:
  # Enable the FPU
  li t0, 0x2000
  csrs mstatus, t0
  csrw	fcsr,x0

  li t0, 0xFFFFFFFF
  fcvt.s.wu	ft0,t0,rtz

  la t0, .fdata0
  fld ft1, 0(t0)

  // Here, ft0 = ft1 = 0xffffffff4f7fffff

  flt.s	t0,ft1,ft0

  li t2, 0x10
  sw t0, 0(t2)

  sw x0, (x0)

infinite_loop:
  jal	x0,infinite_loop

.section ".fdata0","ax",@progbits
  .8byte 0xffffffff4f7fffff
```

### Expected and actual results

We expect `t0 = 0`. I verified this with Spike.
However,  VexRiscv returns `t0 = 1`.

## Notes

This issue has commonalities with #316 . The FPU seems to have a strict ordering between the two floating-point registers, which are in fact equal.

Thanks!
Flavien

Dolu1990, 2023-03-01:
Looking at the wave, the issue is before the flt.s, so either fcvt, either fld. i'm looking a tit.

Dolu1990, 2023-03-01:
Seems like it is due to fcvt storing too much precision bits in the register file.

Dolu1990, 2023-03-01:
https://github.com/SpinalHDL/VexRiscv/issues/317 Seems to fix it.

Thanks :D

flaviens, 2023-03-01:
Cool! Well, thank you for your reactivity! I would not be surprised if it also fixed some other bugs in the batch.

closed 2023-03-01 by flaviens

Dolu1990, 2023-03-01:
So yes, most of the issues were directly related to this :)