SpinalHDL/VexRiscv #318: Issue with consecutive load and store in some circumstances

flaviens, 2023-03-01:
Hi there!

I've been encountering a problem with a load that immediately follows a store to the same address or part of it.
I suspect that it is related to the configuration that I am using.
Can you please tell me 

## Configuration

In the simple SoC that I designed, the memory connected to the CPU responds in the next cycle, i.e., if some memory read request is high in the end of the cycle `n`, then the read response will be high (and the data correct) at the end of the cycle `n+1`.
Therefore, I tried the configuration below, which seems to work fine most of the times but not in the case underlined in this issue.

```
object MyTop{
  def main(args: Array[String]) {
    val report = SpinalConfig(mode = if(args.contains("--vhdl")) VHDL else Verilog).generate{

      //CPU configuration
      val cpuConfig = VexRiscvConfig(
        plugins = List(
          new IBusCachedPlugin(
            resetVector = 0x80000000l,
            compressedGen = false,
            prediction = STATIC,
            historyRamSizeLog2 = 4,
            injectorStage = false,
            config = InstructionCacheConfig(
              cacheSize = 4096*1,
              bytePerLine = 8,
              wayCount = 2,
              addressWidth = 32,
              cpuDataWidth = 32,
              memDataWidth = 64,
              catchIllegalAccess = true,
              catchAccessFault = true,
              asyncTagMemory = false,
              twoCycleRam = false,
              twoCycleCache = false
            ),
            memoryTranslatorPortConfig = MmuPortConfig(
              portTlbSize = 4
            )
          ),
          new DBusCachedPlugin(
            dBusCmdMasterPipe = true,
            dBusCmdSlavePipe = true,
            dBusRspSlavePipe = true,
            config = new DataCacheConfig(
              cacheSize         = 4096*1,
              bytePerLine       = 8,
              wayCount          = 2,
              addressWidth      = 32,
              cpuDataWidth      = 64,
              memDataWidth      = 64,
              catchAccessError  = true,
              catchIllegal      = true,
              catchUnaligned    = true,
              withExclusive = false, // = withsmp
              withInvalidate = false, // = withsmp
              withLrSc = false,
              withAmo = false,
              withWriteAggregation = false
            ),
            memoryTranslatorPortConfig = MmuPortConfig(
              portTlbSize = 4
            )
          ),
          new MmuPlugin(
            virtualRange = _(31 downto 28) === 0xC,
            ioRange      = _(31 downto 28) === 0xF
          ),
          new FpuPlugin(
            externalFpu = false,
            simHalt = false,
            p = FpuParameter(withDouble = true)
          ),
          new DecoderSimplePlugin(
            catchIllegalInstruction = true
          ),
          new RegFilePlugin(
            regFileReadyKind = plugin.SYNC,
            zeroBoot = false
          ),
          new IntAluPlugin,
          new SrcPlugin(
            separatedAddSub = false,
            executeInsertion = true
          ),
          new FullBarrelShifterPlugin,
          new MulPlugin,
          new DivPlugin,
          new HazardSimplePlugin(
            bypassExecute           = true,
            bypassMemory            = true,
            bypassWriteBack         = true,
            bypassWriteBackBuffer   = true,
            pessimisticUseSrc       = false,
            pessimisticWriteRegFile = false,
            pessimisticAddressMatch = false
          ),
          new BranchPlugin(
            earlyBranch = false,
            catchAddressMisaligned = true
          ),
          new CsrPlugin(
            config = CsrPluginConfig(
              catchIllegalAccess = false,
              mvendorid      = null,
              marchid        = null,
              mimpid         = null,
              mhartid        = null,
              misaExtensionsInit = 66,
              misaAccess     = CsrAccess.NONE,
              mtvecAccess    = CsrAccess.NONE,
              mtvecInit      = 0x00000020l,
              mepcAccess     = CsrAccess.READ_WRITE,
              mscratchGen    = false,
              mcauseAccess   = CsrAccess.READ_ONLY,
              mbadaddrAccess = CsrAccess.READ_ONLY,
              mcycleAccess   = CsrAccess.NONE,
              minstretAccess = CsrAccess.READ_ONLY,
              ecallGen       = true,
              ebreakGen      = true,
              wfiGenAsWait   = false,
              ucycleAccess   = CsrAccess.NONE,
              uinstretAccess = CsrAccess.READ_ONLY
            )
          ),
          new YamlPlugin("cpu0.yaml")
        )
      )

      //CPU instanciation
      val cpu = new VexRiscv(cpuConfig)
      cpu
    }
  }
}
```

## Example instances

Here are one instance that fails to read from the write correctly, and two examples where it works, either with a `nop` in between, or with a preliminary floating load.

### Failing instance

```
  .section ".text.init","ax",@progbits
  .globl _start
  .align 2
_start:
  li t1, 0x80003000

  # Store then immediately load
  sw t1, 0(t1)
  lw t2, 0(t1)

  # Write the result to some memory/peripheral to make it clearly visible
  li t0, 0x10
  sw t2, 0(t0)
  sw zero, (zero)

infinite_loop:
  jal	zero, infinite_loop
```
In the snippet above, the final value of `t2` should be `0x80003000` but is `0`.
However, from my limited understanding of the data bus protocol, the correct read data seems to be sampled by the CPU at the right moment.

### Ok instance with nop

```
  .section ".text.init","ax",@progbits
  .globl _start
  .align 2
_start:
  li t1, 0x80003000

  # Store then immediately load after the nop
  sw t1, 0(t1)
  nop
  lw t2, 0(t1)

  # Write the result to some memory/peripheral to make it clearly visible
  li t0, 0x10
  sw t2, 0(t0)
  sw zero, (zero)

infinite_loop:
  jal	zero, infinite_loop
```
This snippet works.

### Ok instance with preliminary floating load

```
  .section ".text.init","ax",@progbits
  .globl _start
  .align 2
_start:
  # Enable the FPU
  li t0, 0x2000
  csrs mstatus, t0
  csrw	fcsr,x0

  la t0, .fdata0 # 0x80002000
  fld ft0, 0(t0)

  li t1, 0x80003000

  # Store then immediately load
  sw t1, 0(t1)
  lw t2, 0(t1)

  # Write the result to some memory/peripheral to make it clearly visible
  li t0, 0x10
  sw t2, 0(t0)
  sw zero, (zero)

infinite_loop:
  jal	zero, infinite_loop

.section ".fdata0","ax",@progbits
  .8byte 0x1122334455667788
```
This snippet works, interestingly. It also works when we replace `fld` with `flw` or `lw`, however the writings to CSRs seem required to make it work, iven with `lw`.

## Waveforms

[Here are the waveforms.](https://github.com/SpinalHDL/VexRiscv/files/10860786/storeload.zip)

I remain at your disposal if you need more info!

Thanks & kind regards
Flavien

Dolu1990, 2023-03-01:
Hi,

I checked the failing instance wave, and looking at what get out of VexRiscv to store t2 to the memory, it seems right : 
[image]
May it be an isssue in the testbench ?

flaviens, 2023-03-01:
Hi @Dolu1990 ,

Thank you for your response! I'm so sorry. I think I did a mistake in the VCD the only time I did not double check :disappointed: .
In the meanwhile, I played a bit more with this sw/lw problem and edited the first message above with the new findings: it seems that the problem is conditioned by the alignment of the instruction pair on 8 bytes.

This works
```
nop
sw
lw
```
but this not
```
# no nop
sw
lw
```
"of course", putting nops between store and load also works.

Please let me know if you need more info!

Dolu1990, 2023-03-01:
Ahhh i found the reason, so basicaly in VexRiscv the d$ memory bus response is a bit tricky.
If you use the non-SMP mode (no memory coherency / fence)  then memory writes should not generate any dbus_rsp.
You currently are in this mode.

Else if using memory coherency / smp, memory write need to produce a write rsp.

(see def withWriteResponse = withExclusive)

That the reason why things goes wrong in that sim (testbench "should" not produce dbus_rsp response on for memory writes)

flaviens, 2023-03-01:
Hi @Dolu1990, correct! Thanks for looking into it! I modified the testbench to comply with what you said and indeed it seems to fix this problem!
Is this requirement documented somewhere?

closed 2023-03-01 by flaviens

Dolu1990, 2023-03-01:
It isn't documented, the idea is more that users would use the already made bridges. For instance, could do : 
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/VexRiscvAxi4LinuxPlicClint.scala#L136

But, it would be best i document it too.

flaviens, 2023-03-01:
I have almost no clue about Spinal but I suppose it would be relatively easy to implement either ignoring the rsp after write or an assertion to check that a rsp never follows a write in this config? Else I'm convinced someone will stumble into it again eventually 😇

Dolu1990, 2023-03-01:
Hmm an assertion to be sure no response come outside a refill would to the work :)