SpinalHDL/VexRiscv #319: Bug report: Writing to a CSR fails in rare microarchitectural conditions

flaviens, 2023-03-03:
Hi there!

Hi there!

I've detected a bug in VexRiscv.

## Brief bug description

In rare microarchitectural conditions, writing to fcsr does not succeed as expected. Instead, the CPU hangs or takes an exception.

## Example instances

I provide two ELF files that differ only by one non-taken branch. The first one (`buggy.elf`) executes this non-taken branch (at `PC=0x80001080`) correctly by not taking it and proceeds, but later the write to the CSR fails (at `PC=0x80098f00`). The second ELF (`success.elf`) does not see this non-taken branch and proceeds directly to the next instruction. The write to the CSR works correctly. 

In [this archive you will find](https://github.com/SpinalHDL/VexRiscv/files/10885411/branchcsr.zip) the...

* Failing ELF
* Succeeding ELF
* Waveforms for failing ELF
* Waveforms for succeeding ELF

You will notice, for example, that the ELF should write to address 0 at the end of the testbench but does not in `buggy.vcd`.

## Notes

I'm really sorry, because of my unfamiliarity with Spinal and the lack of readability of the output Verilog code, I am not able to analyze efficiently the causes of this bug in the codebase. I'm very happy to help if you need more info, though!

The ELF content is essentially setting registers to some debug values, then executing the problematic code, and then trying to dump the values.

Thanks!
Flavien

Dolu1990, 2023-03-03:
Hi ^^
Thanks for the report.
What cpu configuration are you using ?

Dolu1990, 2023-03-03:
Seems like you run some RVC code on a non-RVC config : 
800000a0:       2cd6                    fld     fs9,336(sp)

And also that the config isn't setuped to catch illegal instruction ?

if both of those suposition are right, then it isn't a bug :)

flaviens, 2023-03-03:
Hi @Dolu1990 thank you for your quick answer!
* I'm using the same config as in #318 , in the master branch (where I cherry-picked you fix of #316, but unlikely to be related to this issue).
* It looks that illegal instructions should be caught given this config, but given the next point it does not matter sthat much
* except mistake of mine, the commpressed instructions that you may see in the dump of the ELF is objdump mistakingly trying to interpret embedded data as instructions. You can run a quick trace with spike for instance and see that these "instructions" are never executed. But indeed this compressed instruction is on a speculative path 🙂 suspicious 🧑‍🚒

Dolu1990, 2023-03-04:
Hi ^^
So the issue seems the following : 
- [Speculative] execution of a invalid instruction
- The decoder will see it is illegal but the issue is that the decoded signal which specify if that's for the FPU are not specified (don't care) and that the fork to the FPU is also in the decode stage.
- That fork stuff to the FPU which isn't coherent with the other decoded signals 
- CPU wait forever

I'm now trying to reproduce, the fix should be easy

Dolu1990, 2023-03-05:
Hi, for me, the following change fixed it : 
(FpuPlugin.scala https://github.com/SpinalHDL/VexRiscv/commit/153445ff2145f1a1ee2ed7ccd042a34da0876887)

Thanks for the bug report :)

closed 2023-03-05 by flaviens