SpinalHDL/VexRiscv #320: How to judge the Coremark/Dhrystone finish?

xie-1399, 2023-03-04:
Hi
Now I have this structure of Vexriscv cpu

I try other tests and can find the condition to pass the tests , but I don't know how to judge the Coremark/Dhrystone finish and how to get the results of Coremark/Dhrystone，I'm Feeling confused，Would you do me a favor？Thank You!

Dolu1990, 2023-03-05:
When the benchmark return from main it is finished i would say.
Then for the results, they are normaly printed into the terminal.
You are in a simulation ?

xie-1399, 2023-03-05:
Hi
Thank You for your answer ,  But I can't find the Assembly language of the coremark.bin，could you tell me how to compile coremark.bin？

Dolu1990, 2023-03-07:
Hi,
Normaly, you have a .elf right ?
you  can use objdump to convert it into bin : 
https://stackoverflow.com/questions/49680382/objcopy-elf-to-bin-file

xie-1399, 2023-03-07:
Hi
But I just find the coremark.bin，I don’t find the hex file and asm file of the coremark.bin

Dolu1990, 2023-03-08:
Ahh mybad, i missreaded.
So those too can all be generated from the .elf

See the following makefile rules : 

```make
%.hex: %.elf
	$(RISCV_OBJCOPY) -O ihex $^ $@

%.bin: %.elf
	$(RISCV_OBJCOPY) -O binary $^ $@

%.v: %.elf
	$(RISCV_OBJCOPY) -O verilog $^ $@

%.asm: %.elf
	$(RISCV_OBJDUMP) -S -d $^ > $@
```

xie-1399, 2023-03-10:
Hi
In fact , I don't know how to link the coremark at 80000000，So I can't get the coremark.elf，I hope to konw how did you compile the coremark.bin

Dolu1990, 2023-03-10:
Hoo, i compiled that coremark soooooo long ago.
Maybe, check https://github.com/SpinalHDL/NaxSoftware/tree/3c1a611b3fa65974d4a93b760918cb9e94748434/baremetal/coremark

It isn't for VexRiscv, so would need some adaptation in the software.

xie-1399, 2023-03-11:
Thank You！I solved my problem

closed 2023-03-11 by xie-1399