SpinalHDL/VexRiscv #321: Bug report: single-precision multiplication followed by conversion sets wrong bits in the mantissa

flaviens, 2023-03-04:
Hi there!

I've detected a bug in VexRiscv.

## Brief bug description

A single-precision multiplication followed by a single-to-double precision conversion sets wrong bits in the mantissa.
This suggests that 5f67075 did not cover all cases. Consequences are likely to be similar to the bugs that 5f67075 intended to fix.

I ran this code on master (b29eb54) where I cherry-picked the fix 5f67075, with the configuration described in #318 .

## Example instance

### Snippet

Here is an example RISC-V (rv32imfd) snippet:
```
  .section ".text.init","ax",@progbits
  .globl _start
  .align 2
_start:
  # Enable the FPU
  li t0, 0x2000
  csrs mstatus, t0
  csrw	fcsr,x0

  la t0, .fdata0
  fld fs1, (t0)

  fmul.s fa5, fs1, fs1
  fcvt.d.s ft7, fa5

  li t0, 0x18
  fsd ft7, 0(t0)

  sw x0, (x0)

infinite_loop:
  jal	x0,infinite_loop

.section ".fdata0","ax",@progbits
  .8byte 0xffffffff1cb31090 # fs1
```

### Waveform

[Here is a waveform.](https://github.com/SpinalHDL/VexRiscv/files/10888359/fmulfcvt.zip)

### Expected and actual results

We expect `ft7 = 0x373f500000000000`. I verified this with Spike.
However,  VexRiscv returns `ft7 = 0x41e000620a000000` .

Thanks!
Flavien

Dolu1990, 2023-03-05:
Thanks :D
I will check all of this + others issues asap
By the way, what are you using to generate the test vectors ?
Are you specificaly targetting the fpu tests ?

flaviens, 2023-03-05:
Sounds good! I'm using a new technique, which does not focus on FPUs, but the technique lets us observe that FPUs are generally particularly buggy. Once it is is public, I'll be happy to share it and support so you can use it yourself directly :slightly_smiling_face:.

Dolu1990, 2023-03-08:
see #322 resolution

commit 1179c655 (2023-03-08): Fix #321 #322 #333 FPU precision removal
closed 2023-03-08 by Dolu1990