SpinalHDL/VexRiscv #322: Bug report: dependent arithmetic/muldiv single-precision operations cause wrong output by more than +1 or -1

flaviens, 2023-03-04:
Hi there!

I've detected a bug in VexRiscv, similar to #315 but the mismatch may be larger than +1 or -1.

## Brief bug description

A pair of dependent single-precision floating arithmetic/muldiv operations (not necessarily immediately following each other), provides wrong results in some cases, differing from the expectation by potentially a magnitude above 1.

I ran this code on master (b29eb54) where I cherry-picked the fix 5f67075, with the configuration described in #318 .

## Snippet

Here is an example RISC-V (rv32imfd) snippet:
```
  .section ".text.init","ax",@progbits
  .globl _start
  .align 2
_start:
  # Enable the FPU
  li t0, 0x2000
  csrs mstatus, t0
  csrw	fcsr,x0

  la t0, .fdata0
  fld fa2, (t0)
  la t0, .fdata1
  fld ft7, (t0)

  fdiv.s	fs0,ft7,fa2,rmm
  fmul.s	fa2,fs0,fa2,rdn

  li t0, 0x18
  fsd fa2, 0(t0)

  sw x0, (x0)

infinite_loop:
  jal	x0,infinite_loop

.section ".fdata0","ax",@progbits
  .8byte 0xffffffff5f4135ef # fa2
.section ".fdata1","ax",@progbits
  .8byte 0xffffffff98e7d58f # ft7
```

### Expected and actual results

We expect `fa2= 0xffffffff98e7b3ae`. I verified this with Spike.
However,  VexRiscv returns `fa2 = 0xffffffff98e7d58f`.

Thanks!
Flavien

Dolu1990, 2023-03-08:
Thanks for the report ^^
That case and the two #321 #323 are now fixed.

They were all related to not removing enough precision bits from the subnormal values.
Would there be a way for me to contact you with a call ?

closed 2023-03-08 by Dolu1990

flaviens, 2023-03-08:
Hi @Dolu1990 thanks for the fixes 🚀 ! I'll contact you out of band.