SpinalHDL/VexRiscv #324: How can I

ZhengJuzhong666, 2023-03-13:


Dolu1990, 2023-03-14:
Hi,

Hmm i'm not exactly sure of your actual setup. can you share the modifications ?
Also, if i understand well, the error pop up in Line 313 at FpuPlugin.scala, while you didn't modified that file right ?

ZhengJuzhong666, 2023-03-14:
Thank you!

I modified this part of the code.
Line 128 is the original code. The commented code above is the part I want to modify. I want to instead line 128 with commented code. 
[image]
I understand this is the code used to pass signals in the CPU pipeline. But I don't know much about the specific design of the FPU unit, so it's not clear to me why this variable won't be instantiated as a register.

Dolu1990, 2023-03-21:
Hi, sorry for the delay, 
> I understand this is the code used to pass signals in the CPU pipeline.

yes that's right.
So i think the issue is that it doesn't like the "assignFromBits being used here.
:= isn't working ? else, assignFrom ?