SpinalHDL/VexRiscv #325: DBusCachedPlugin and Peripheral

joeljeanmonod-rgb, 2023-03-29:
Hi,

First thank for this amazing CPU.

I have a question about the DBusCachedPlugin. 
For the context, the configuration that i want is two AXI bus one for the iBus and one for the dBus. The AXI (ibus) is connected to the flash and the second  AXI (dbus) is connected to the RAM core. The iBus and dBus are cached.

My question is if i want to add a timer for exemple to the dbus, how can i say i don't want the dbus to cache the registers of the timer ?

Dolu1990, 2023-03-29:
Hi,

Thanks :)

So if you don't have a MMU, you have that  in the StaticMemoryTranslatorPlugin parameters : 
https://github.com/SpinalHDL/VexRiscv/blob/f0bb6e94e4be6eeccade29210263c5ff305fc227/src/main/scala/vexriscv/demo/Briey.scala#L104

the _ symbole represent the address.
a longer form is : 
ioRange      = addr => addr (31 downto 28) === 0xF

Dolu1990, 2023-03-29:
ioRange meaning => all the address which aren't cached.

joeljeanmonod-rgb, 2023-03-30:
thank you very much

closed 2023-03-30 by joeljeanmonod-rgb