SpinalHDL/VexRiscv #328: Breakpoint exceptions cannot be delegated to Supervisor

xobs, 2023-04-10:
The `medeleg` register allows users to delegate exceptions to a lower privilege level. However, not all `medeleg` bits are wired up in Vexriscv. In particular, `Breakpoint` is not wired up, which means a User program can trap to Machine mode by issuing `ebreak`. Additionally, this means that debuggers cannot be implemented without having a small program that runs in machine mode.

https://github.com/SpinalHDL/VexRiscv/blob/c52433575dec04f10063b2fd7cebd0545c8b1be9/src/main/scala/vexriscv/plugin/CsrPlugin.scala#L1005-L1008

How difficult would it be to allow Breakpoints to be delegated like all other exception types?

Dolu1990, 2023-04-10:
Hi,

Yes right ^^
It should be as simple as doing : 

```scala
      val medeleg = supervisorGen generate new Area {
        val IAM, IAF, II, BP, LAM, LAF, SAM, SAF, EU, ES, IPF, LPF, SPF = RegInit(False)
        val mapping = mutable.LinkedHashMap(0 -> IAM, 1 -> IAF, 2 -> II, 3 -> BP, 4 -> LAM, 5 -> LAF, 6 -> SAM, 7 -> SAF, 8 -> EU, 9 -> ES, 12 -> IPF, 13 -> LPF, 15 -> SPF)
      }
```

I will test it an put it in dev :)

Dolu1990, 2023-04-10:
ebreak medeleg added in dev, tested, seems to work well :)

closed 2023-04-10 by Dolu1990

xobs, 2023-04-10:
Thanks for that! I'll have @bunnie test it on Precursor.

commit d966c4ef (2023-04-10): fix #328 medeleg EBREAK added