SpinalHDL/VexRiscv #329: CPU Configuration with PMP Plugin

seceng-jan, 2023-04-13:
Hi!

I am trying to instantiate the PmpPlugin in a CPU configuration that boots Linux. Therefore, I modified the [VexRiscvSmpCluster](https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/smp/VexRiscvSmpCluster.scala) configuration and added 
```
new PmpPlugin(
  regions = 16,
  granularity = 32,
  ioRange = _(31 downto 28) === 0xf
),
```
When I run sbt, the following error occurs:
```
[info] **********************************************************************************************
[info] [Warning] Elaboration failed (0 error).
[info]           Spinal will restart with scala trace to help you to find the problem.
[info] **********************************************************************************************
[info] [Progress] at 0.428 : Elaborate components
[error] Exception in thread "main" java.lang.AssertionError: assertion failed: ??? vexriscv.MemoryTranslator
[error]         at scala.Predef$.assert(Predef.scala:170)
[error]         at spinal.core.package$.assert(core.scala:500)
[error]         at vexriscv.Pipeline$class.service(Pipeline.scala:26)
[error]         at vexriscv.VexRiscv.service(VexRiscv.scala:123)
[error]         at vexriscv.plugin.IBusCachedPlugin.setup(IBusCachedPlugin.scala:114)
[error]         at vexriscv.plugin.IBusCachedPlugin.setup(IBusCachedPlugin.scala:28)
[error]         at vexriscv.Pipeline$$anonfun$build$2.apply(Pipeline.scala:47)
[error]         at vexriscv.Pipeline$$anonfun$build$2.apply(Pipeline.scala:47)
[error]         at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error]         at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error]         at vexriscv.Pipeline$class.build(Pipeline.scala:47)
[error]         at vexriscv.VexRiscv.build(VexRiscv.scala:123)
[error]         at vexriscv.Pipeline$$anonfun$1.apply$mcV$sp(Pipeline.scala:161)
[error]         at spinal.core.Component$$anonfun$prePop$1.apply(Component.scala:180)
[error]         at spinal.core.Component$$anonfun$prePop$1.apply(Component.scala:178)
[error]         at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error]         at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error]         at spinal.core.Component.prePop(Component.scala:178)
[error]         at spinal.core.Component.postInitCallback(Component.scala:187)
[error]         at vexriscv.VexRiscvBmbGenerator$$anonfun$10$$anon$2.<init>(VexRiscvBmbGenerator.scala:111)
[error]         at vexriscv.VexRiscvBmbGenerator$$anonfun$10.apply(VexRiscvBmbGenerator.scala:99)
[error]         at vexriscv.VexRiscvBmbGenerator$$anonfun$10.apply(VexRiscvBmbGenerator.scala:99)
[error]         at spinal.core.fiber.package$$anonfun$1.apply$mcV$sp(package.scala:10)
[error]         at spinal.core.fiber.AsyncThread$$anonfun$1.apply$mcV$sp(AsyncThread.scala:59)
[error]         at spinal.core.fiber.EngineContext$$anonfun$newJvmThread$1.apply$mcV$sp(AsyncCtrl.scala:39)
[error]         at spinal.sim.JvmThread.run(SimManager.scala:51)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 8 s, completed Apr 13, 2023, 9:45:13 AM
```

The same happens with the GenFull configuration. I assmue that this is related to the memoryTranslatorPortConfig but I am not sure what to put there. Maybe it also relates to #307, I tried the modifications there but same result. Can you help me out?

Dolu1990, 2023-04-14:
Hi,

MMU + PMP at the same time isn't supported in VexRiscv, it is either one or the other or none

seceng-jan, 2023-04-14:
Thank you! Do you happen to know if anyone has been successful in getting the keystone TEEs to work on vexriscv? I suppose it is possible to compile the Linux kernel without MMU support, but it will probably be very limited and I am not sure if Keystone will be compatible.

Dolu1990, 2023-04-17:
Ahh i have no idea, never heared of keystone TEEs.
Also, never tried linux without the MMU.

seceng-jan, 2023-04-19:
Okay thank you! :-) I'll try to get it working somehow.

closed 2023-04-19 by seceng-jan