SpinalHDL/VexRiscv #332: Unspecified value parameter jtagHeaderIgnoreWidth

FaizanAhmad626, 2023-04-16:
I am running into this error while generating core from VexRiscv.scala for ztachip. There is some new parameter "jtagHeaderIgnoreWidth" whose value is not declared in this file [https://github.com/ztachip/ztachip/blob/master/HW/examples/GHRD/MyVexRiscv.scala#L206](https://github.com/ztachip/ztachip/blob/master/HW/examples/GHRD/MyVexRiscv.scala#L206).
[image]

What to do about it?

Dolu1990, 2023-04-17:
Hi,

Hmm you can do fromJtagInstructionCtrl(ClockDomain(tap.TCK), 0)

That should be good then.

FaizanAhmad626, 2023-04-23:
Okay that issue is resolved but now I am having these errors:
[image]

ztachip, 2023-04-23:
I notice that my run has an older spinalHDL version v1.6.0 while yours is v1.8.1
Probably there is some new syntax on latest release.
I will need to upgrade to latest spinalHDL release to fix this.
Meanwhile, if you change the MaxPendingWrite in AXI crossbar, that should do the same.

[image]<https://user-images.githubusercontent.com/111587195/233824688-493e8685-4ecf-4830-b2ac-d35ccbb521ef.png>

FaizanAhmad626, 2023-04-23:
Okay.

FaizanAhmad626, 2023-04-28:
Now I added 'C' extension to [MyVexRiscv.scala](https://github.com/ztachip/ztachip/blob/master/HW/examples/GHRD/MyVexRiscv.scala) by adding this line here:
[image]

Not sure if its correct or not. But its still giving those errors which @ztachip noticed is due to running newer version of SpinalHDL.

Dolu1990, 2023-04-28:
What VexRiscv github branch / SpinalHDL is used ?

FaizanAhmad626, 2023-04-28:
I am using SpinalHDL v 1.8.1 whereas this scala file [https://github.com/ztachip/ztachip/blob/master/HW/examples/GHRD/MyVexRiscv.scala](https://github.com/ztachip/ztachip/blob/master/HW/examples/GHRD/MyVexRiscv.scala) is using SpinalHDL v 1.6.0

FaizanAhmad626, 2023-04-29:
> What VexRiscv github branch / SpinalHDL is used ?

I am using master branch for building VexRiscv from scala. But not sure about the branch of MyVexRiscv.

Dolu1990, 2023-05-01:
Ahhh i think that the issue, master VexRiscv is meant to be used with a release of SpinalHDL, not dev.

So i think you can fix that jtagHeaderIgnoreWidth to 0, that should work fine.
Or switch to vexriscv dev aswell

FaizanAhmad626, 2023-05-01:
I tried master version of SpinalHDL and didn't work.

Then I tried dev version of VexRiscv and still didn't work.

I may be making a mistake somewhere may be. I tried fullGen version and it compiled successfully but I don't know why its giving errors in MyVexRiscv version.

ztachip, 2023-05-01:
Can you try the example scala that comes with VexRiscv

FaizanAhmad626, 2023-05-02:
I've tried this example:
sbt "runMain vexriscv.demo.GenFull"
And it generated the Verilog file successfully.

ztachip, 2023-05-03:
Try copy MyVexRiscv.scala to same folder where you find GenFull.scala and try to compile again
Make sure there is just one copy of MyVexRiscv.scala in the VexRiscv folder since the compiler would look for all MyVexRiscv.scala from all folders under VexRiscv folder.

FaizanAhmad626, 2023-05-03:
I tried and again failed. I think there is some branch issue may be. Which branch did you use? I tried master branch and it failed to compile MyVexRiscv.scala. The dev branch even failed to compile GenFull.scala I don't know why.

[image]
And by looking at these errors I concluded that there is some issue related to the libraries you imported for MyVexRiscv.scala which are not imported in the GenFull.scala file. This is due to the extra functionalities in MyVexRiscv version so may be some extra per-requisites must also be required. Right now I am just cloning VexRiscv from GitHub and installed its per-requisites only.

I don't know if this will be emailed to you or not so I am mentioning you in this comment @ztachip

ztachip, 2023-05-03:
I will try later a fresh installation of latest VexRiscv in another PC.
Probably I got a mix version for some reason.
Will let you know

[image]<https://user-images.githubusercontent.com/111587195/235888099-8f521c32-691c-460c-92a8-0f02b0e2b83b.png>

FaizanAhmad626, 2023-05-03:
Okay.

closed 2023-05-21 by FaizanAhmad626