SpinalHDL/VexRiscv #333: Application userspace/hello_world_user in SoC LiteX VexRiscv

KevinQhv, 2023-04-17:
Hi, 

I try an application provided by Zephyr `userspace/hello_world_user` on my SoC that i created with LiteX.

```
Configuration :
- OS : Ubuntu 22.04 LTS
- Board : Digilent Arty A7 100T
- LiteX Version 2022.08
- CPU : VexRiscv Variant : Secure, Linux, Full, Lite
- OS : ZephyrOS lastest 3.1.0
```

I tried the application with several variants Full, Lite, Linux and Secure for the VexRiscv CPU. But, I don't enter the thread in user mode. The `k_is_user_context() ` function does not return 1 for user mode for all variants. 

```
ASSERTION FAIL [k_is_user_context()] @ WEST_TOPDIR/zephyr/samples/userspace/hello_world_user/src/main.c:19
        User mode execution was expected
E:      a0: 00000004    t0: 00000000
E:      a1: 0000001c    t1: 00000000
E:      a2: 400053c8    t2: ffffffff
E:      a3: 40006d84    t3: 00000020
E:      a4: 00000000    t4: 00000023
E:      a5: f0003000    t5: ffffffff
E:      a6: 20000000    t6: ffffffff
E:      a7: 00000000
E:      ra: 40000ac0
E:    mepc: 40001da0
E: mstatus: 00001880
E: 
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x400063d0 (unknown)
E: Halting system
```

So, i got an ASSERTION FAIL with the function `__ASSERT(k_is_user_context()` but, I want to enter the user mode. The Linux, Full and Lite variants have three privilege modes.

Thank you for your help

Dolu1990, 2023-04-17:
What DTS did you used to compile the zephyr ? (nothing pop up in sim)
Is there a easy way for me to recompile things ?

Dolu1990, 2023-04-17:
(i'm trying to repoduce the issue in simulation, without all the arty related stuff, so using ./litex_sim)

Dolu1990, 2023-04-17:
one thing i just thinked about, you may have more success with vexriscv_smp cpu variant, as it enable more CSR features

KevinQhv, 2023-04-17:
The DTS file generated by the compilation of the SoC following the Zephyr Doc liteX VexRiscv 
[https://docs.zephyrproject.org/latest/boards/riscv/litex_vexriscv/doc/index.html#litex-vexriscv](url) with the command for the board Arty A7-100T : 
`./litex-boards/litex_boards/targets/digilent_arty.py --variant=a7-100 --cpu-variant=secure --build --timer-uptime --csr-json csr.json --load`
and for genrated overlay.dts : 
`./litex/litex/tools/litex_json2dts_zephyr.py --dts overlay.dts --config overlay.config csr.json`

And I just tested the application on the vexriscv_smp processor variant Linux and I have a different problem: 
`
E: 
E:  mcause: 2, Illegal instruction
E:   mtval: bc002773
E:      a0: 00000001    t0: 00000800
E:      a1: 4000816e    t1: 0000006a
E:      a2: 00000006    t2: 0000007f
E:      a3: 00000007    t3: 0000007a
E:      a4: 40005d34    t4: 00000074
E:      a5: 40002b2c    t5: 00000009
E:      a6: 00000068    t6: 0000002a
E:      a7: 0000004c
E:      ra: 40002b3c
E:    mepc: 40002284
E: mstatus: 00001900
E: 
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x40007478 (unknown)
E: Halting system
`

Dolu1990, 2023-04-17:
Ahh thanks for the info :D
Got the overlay.dts. 
Which version of zephyr did you used ? how do you compile it ? (didn't used zephyr since years)

About the crash you get on vexriscv_smp, it is because you use the zephyr compiled for the vexriscv non smp version, which has a different interrupt design. (mtval: bc002773)

You need to recompile zephyr with the vexriscv_smp headers

KevinQhv, 2023-04-17:
I use this configuration : 
```
Configuration :
- OS : Ubuntu 22.04 LTS
- Board : Digilent Arty A7 100T
- LiteX Version 2022.08
- CPU : VexRiscv Variant : Secure, Linux, Full, Lite
- OS : ZephyrOS lastest 3.1.0
```
Once the DTS file is created in the zephyr folder : 
`west build -b litex_vexriscv samples/userspace/hello_world_user  -DDTC_OVERLAY_FILE=~/zephyrproject/PATHTO/overlay.dts`

And for testing : 
`litex_term  /dev/ttyUSB2 --kernel ../zephyr/build/zephyr/zephyr.bin`

I don't see the recompile zephyr with the vexriscv_smp headers, 
Need to create a new board for vexriscv_smp? or, just add a configuration option for vexriscv ?

Dolu1990, 2023-04-18:
>  I don't see the recompile zephyr with the vexriscv_smp headers,
Need to create a new board for vexriscv_smp? or, just add a configuration option for vexriscv ?

So, the thing is, if you got 0xbc002773 instruction in you binary, it mean you had something litex specific in the zephyr.
Did you use some zephyr fork or something ?

(all of this is a flow i do not know)

Ultimately, the thing i need to debug the issue, is to be able to run a simulation of the system.
Did you used "litex_sim" by the past ?

KevinQhv, 2023-04-18:
To create the Overlay.dts file, I used a python script `litex_json2dts_zephyr.py` and zephyr takes this file to build the application.
I did not use `litex_sim` with a zephyr application. I just tested it on a board.

KevinQhv, 2023-07-04:
the answer is in this issue #https://github.com/zephyrproject-rtos/zephyr/issues/56789

closed 2023-07-04 by KevinQhv