SpinalHDL/VexRiscv #334: Bug report: VexRiscv should raise an illegal instruction exception when reading unsupported CSRs

flaviens, 2023-04-27:
Hi there!

### Bug description

I think I found a bug not described in the issues and PRs in this repo.
When accessing unsupported CSRs, VexRiscv should raise an illegal instruction, but instead proceeds without throwing an exception.

### Example snippet

This snippet should write 1 to 0x10 but writes 0 instead.

```
  .section ".text.init","ax",@progbits
  .globl _start
  .align 2
_start:

  la 	  ra, trap_handler
  csrrw	zero,mtvec,ra
  csrr	zero,ustatus

  li a0, 0x10
  li t0, 0

  sd t0, (a0)
  sd zero, (zero)

infinite_loop0:
  j infinite_loop0

.section ".trap_handler","ax",@progbits
trap_handler:

  li a0, 0x10
  li t0, 1

  sd t0, (a0)
  sd zero, (zero)

infinite_loop1:
  j infinite_loop1
```

Thanks!

Dolu1990, 2023-04-27:
Hi,
Welcome back <3 XD

So, i just tried localy, to see if ustatus access trap, and on my setup they do.

Can you send me your wave file + vexriscv.v ^^ ?

thanks :D
Charles

flaviens, 2023-04-27:
Hi Charles,

Thanks :)
I'm very sorry, it was a mistake on my side. If my config has
```
      CsrPluginConfig(
        catchIllegalAccess = false,
```
then I should not be surprised by this behaviour.

Sorry for the inconvenience!

closed 2023-04-27 by flaviens

Dolu1990, 2023-04-27:
no worries :)