SpinalHDL/VexRiscv #336: Bug report: Operations on floating-point registers should raise an exception when FS=0

flaviens, 2023-04-27:
Hi there!

### Bug description

I think I found a bug not described in the issues and PRs in this repo, but very similar and probably very related to #335 .
When FS is 0, operations on floating-point registers should raise an illegal instruction exception, according to the table `Table 3.4: FS, VS, and XS state transitions.` of the riscv privileged specification.
However, it seems that currently VexRiscv does not raise an exception in this scenario.
Let me know if it's just a question of configuration.

### Example snippet

This snippet should write 1 to 0x10 but writes 0 instead.

```
  .section ".text.init","ax",@progbits
  .globl _start
  .align 2
_start:

  li a0, 0x10

  la 	  ra, trap_handler
  csrrw	zero,mtvec,ra

  # Clear FS
  li t0, 0x2000
  csrc mstatus, t0

  # Check that FS is effectively 2'00
  # csrr	t0,mstatus
  # srli  t0, t0, 13
  # andi  t0, t0, 3
  # sw t0, (a0)

  fadd.s f0, f0, f0 # Here an exception should be raised
  li t0, 0

  sw t0, (a0)
  sw zero, (zero)

infinite_loop0:
  j infinite_loop0

.section ".trap_handler","ax",@progbits
trap_handler:

  li t0, 1

  sw t0, (a0)
  sw zero, (zero)

infinite_loop1:
  j infinite_loop1
```

Configuration:
```
  def main(args: Array[String]) {
    val report = SpinalConfig(mode = if(args.contains("--vhdl")) VHDL else Verilog).generate{

      //CPU configuration
      val cpuConfig = VexRiscvConfig(
        plugins = List(
          new IBusCachedPlugin(
            resetVector = 0x80000000l,
            compressedGen = false,
            prediction = STATIC,
            historyRamSizeLog2 = 4,
            injectorStage = false,
            config = InstructionCacheConfig(
              cacheSize = 4096*1,
              bytePerLine = 8,
              wayCount = 2,
              addressWidth = 32,
              cpuDataWidth = 32,
              memDataWidth = 64,
              catchIllegalAccess = true,
              catchAccessFault = true,
              asyncTagMemory = false,
              twoCycleRam = false,
              twoCycleCache = false
            ),
            memoryTranslatorPortConfig = MmuPortConfig(
              portTlbSize = 4
            )
          ),
          new DBusCachedPlugin(
            dBusCmdMasterPipe = true,
            dBusCmdSlavePipe = true,
            dBusRspSlavePipe = true,
            config = new DataCacheConfig(
              cacheSize         = 4096*1,
              bytePerLine       = 8,
              wayCount          = 2,
              addressWidth      = 32,
              cpuDataWidth      = 64,
              memDataWidth      = 64,
              catchAccessError  = true,
              catchIllegal      = true,
              catchUnaligned    = true,
              withExclusive = false, // = withsmp
              withInvalidate = false, // = withsmp
              withLrSc = false,
              withAmo = false,
              withWriteAggregation = false
            ),
            memoryTranslatorPortConfig = MmuPortConfig(
              portTlbSize = 4
            )
          ),
          new MmuPlugin(
            virtualRange = _(31 downto 28) === 0xC,
            ioRange      = _(31 downto 28) === 0xF
          ),
          new FpuPlugin(
            externalFpu = false,
            simHalt = false,
            p = FpuParameter(withDouble = true)
          ),
          new DecoderSimplePlugin(
            catchIllegalInstruction = true
          ),
          new RegFilePlugin(
            regFileReadyKind = plugin.SYNC,
            zeroBoot = false
          ),
          new IntAluPlugin,
          new SrcPlugin(
            separatedAddSub = false,
            executeInsertion = true
          ),
          new FullBarrelShifterPlugin,
          new MulPlugin,
          new DivPlugin,
          new HazardSimplePlugin(
            bypassExecute           = true,
            bypassMemory            = true,
            bypassWriteBack         = true,
            bypassWriteBackBuffer   = true,
            pessimisticUseSrc       = false,
            pessimisticWriteRegFile = false,
            pessimisticAddressMatch = false
          ),
          new BranchPlugin(
            earlyBranch = false,
            catchAddressMisaligned = true
          ),
          new CsrPlugin(
            config = CsrPluginConfig(
              catchIllegalAccess = true,
              mvendorid      = 0,
              marchid        = 0,
              mimpid         = 0,
              mhartid        = 0,
              misaExtensionsInit = 66,
              misaAccess     = CsrAccess.NONE,
              mtvecAccess    = CsrAccess.READ_WRITE,
              mtvecInit      = 0x00000020l,
              mepcAccess     = CsrAccess.READ_WRITE,
              mscratchGen    = false,
              mcauseAccess   = CsrAccess.READ_ONLY,
              mbadaddrAccess = CsrAccess.READ_ONLY,
              mcycleAccess   = CsrAccess.NONE,
              minstretAccess = CsrAccess.READ_ONLY,
              ecallGen       = true,
              ebreakGen      = true,
              wfiGenAsWait   = false,
              ucycleAccess   = CsrAccess.NONE,
              uinstretAccess = CsrAccess.READ_ONLY
            )
          ),
          new YamlPlugin("cpu0.yaml")
        )
      )

      val cpu = new VexRiscv(cpuConfig)
      cpu
    }
  }
```
[Waveform.]https://github.com/SpinalHDL/VexRiscv/files/11344155/sim.zip)

Thanks!

flaviens, 2023-04-27:
Fixed by https://github.com/SpinalHDL/VexRiscv/commit/6be1531d366f30b306c381a0b455cbe1f7b4478a

closed 2023-04-27 by flaviens

Dolu1990, 2023-04-27:
<3