SpinalHDL/VexRiscv #339: Questions about the memory of the Murax SoC and the VexRiscv in general

joeljeanmonod-rgb, 2023-05-05:
Hi,

I have a few questions about how the memory map work with Murax Soc and the VexRiscv in general.

My first question is about how the processor starts. Of what i understand the sequence is as follow :

1. Start at the adress of the `resetVector  = 0x80000000l`
2. Read the instructions of the `crt.S` which are at the adress `0x80000000l`
4. The `crt.S` enable the interrupts for the timer and the uart and then jump to the `main` function
3. Start the the main function in the `main.c`

is that correct ?

My second question is about the memory map. I have a few questions about it :

1. Is the processor using a part of the memory map for internal use ?
2. if yes, where can i find the memory map of the processor for internal use ?
3. if no,  does that mean that the memory map entirely available for the user ?  Can for exemple put the RAM at the adress `0x00000000l` directly with the `resetVector  = 0x00000000l` ?

thanks in advance for your response.

Dolu1990, 2023-05-05:
HI ^^

> The crt.S enable the interrupts for the timer and the uart and then jump to the main function

Not exactly, it enable timer and external interrupts in the core, the uart interrupt is connected to a interrupt controller which goes to that external interrupt.

> Is the processor using a part of the memory map for internal use ?

No

> if no, does that mean that the memory map entirely available for the user ? 

Yes

> Can for exemple put the RAM at the adress 0x00000000l directly with the resetVector  = 0x00000000l ?

Yes, but overall i would say the best is to put nothing at 0x00000000 (so null pointer access will not access memory)

joeljeanmonod-rgb, 2023-05-07:
Thank you so much

closed 2023-05-07 by joeljeanmonod-rgb