SpinalHDL/VexRiscv #340: Bug report: VexRiscv raises an illegal instruction when writing to mcycle, mcycleh, minstret and minstreth

flaviens, 2023-05-05:
Hi there!

### Bug description

I think I found a bug not described in the issues and PRs in this repo.
When writing mcycle, mcycleh, minstret and minstreth, VexRiscv raises an illegal instruction instead of writing these registers.
The privileged spec (both 1.10 and the most recent version) clearly states that these CSRs are software-writable.

### Example snippet

This snippet should store `0xc001` to the address `0x10`, but stores `0xbaad` instead.

```
  .section ".text.init","ax",@progbits
  .globl _start
  .align 2
_start:

  li a0, 0x10

  la 	  t0, trap_handler
  csrw  mtvec, t0

  csrw  mcycleh, zero

  sw t0, (a0)
  li t0, 0xc001
  sw t0, (a0)
  sw zero, (zero)

infinite_loop0:
  j infinite_loop0

# .section ".trap_handler","ax",@progbits
trap_handler:

  csrr  t0, mcause
  sw t0, (a0)
  csrr  t0, mepc
  sw t0, (a0)
  li t0, 0xbaad
  sw t0, (a0)
  sw zero, (zero)

infinite_loop1:
  j infinite_loop1
```

Thanks!

Dolu1990, 2023-05-05:
Hi ^^

Which CSR config do you use ? 

Are : 
    mcycleAccess        = CsrAccess.READ_WRITE,
    minstretAccess      = CsrAccess.READ_WRITE,
    ?
    
    Often they are read only to save area

flaviens, 2023-05-05:
That means it's soon time to go in week end...
Thanks for your reactivity and sorry ^^'

closed 2023-05-05 by flaviens

Dolu1990, 2023-05-05:
No worries :)