SpinalHDL/VexRiscv #341: Bug report: VexRiscv overcounts the retired instructions

flaviens, 2023-05-05:
Hi there!

### Bug description

I think I found a bug not described in the issues and PRs in this repo.
`minstret` seems to overcount.

### Example snippet

The first write (corresponding to `x1`, freshly read from `minstret`) to address `0x10` should be `0x29`, according to spike. It is, however, `0x2a` according to Vexriscv.

[Here is an archive containing both the ELF and the waveforms.](https://github.com/SpinalHDL/VexRiscv/files/11409563/overcount.zip)

Thanks!

Dolu1990, 2023-05-05:
Hmm it is quite possibly due to csrrw	x0,minstret,x0  which once it get commited increase minstret ? While on spike it increase first and then set to zero ?

flaviens, 2023-05-07:
Yes exactly. I simplified the example snippet above.
Interestingly, the expected behavior is very clearly stated in the 1.12 privileged spec:
> Any CSR write takes effect after the writing instruction has otherwise completed.
However, this explicit mention is absent from the 1.10 version, but this clarification is also omitted from the changelog in the 1.12 privileged spec, hence I suppose that the behavior better explained in 1.12 is also the one expected from 1.10.
It's probably not the most critical mismatch Vexriscv got in its life, though :disguised_face:  .

Dolu1990, 2023-05-09:
Hmmm, for now i kind of thinking a "won't fix" is ok :)

Thanks anyway :D

flaviens, 2023-05-09:
Yeah I understand :)

closed 2023-05-09 by flaviens